NicoHood/HID #7: Possible errors in the description of the installation

bgma, 2015-01-08:
Hi,
I'm running ARDUINO 1.6.0rc2 on W7 pro SP1

In the description it says, "At the moment You have to move the cores/HID folder intoyour Arduino installation arduino-1.6.0/hardware/Arduino/AVR/cores/HID Because of a bug in the IDE."

First i renamed the original folder "\Arduino\hardware\arduino\avr\cores\arduino\" then created a new folder : "\Arduino\hardware\arduino\avr\cores\arduino\" then move the contents of "cores/hid" to: "\Arduino\hardware\arduino\avr\cores\arduino\" to compile a blank sketch.

[image]

/BG

NicoHood, 2015-01-08:
You have to put it into cores/hid so you dont touch the cores/arduino at all.

[image]

bgma, 2015-01-08:
Now i have restored the original, but the question is why it doesn't work?
I get the folowing error:

C:\Arduino\hardware\arduino\avr\cores\arduino\USBCore.cpp: In member function 'void USBDevice_::attach()':
C:\Arduino\hardware\arduino\avr\cores\arduino\USBCore.cpp:662:2: error: 'UHWCON' was not declared in this scope
  UHWCON = 0x01;      // power internal reg
  ^
C:\Arduino\hardware\arduino\avr\cores\arduino\USBCore.cpp:677:26: error: 'OTGPADE' was not declared in this scope
  USBCON = ((1<<USBE)|(1<<OTGPADE)); // start USB clock
                          ^
Error compiling.

[image]
Ihave selected HoodLoader2 16u2 as board

NicoHood, 2015-01-08:
HoodLoader2 needs a new board definition file. I will update HoodLoader2 when HID Project 2.1 is ready. You are not able to use HoodLoader2 right now. I am currently working on that, i may upload a new dev branch this evening for HL2 as well.

NicoHood, 2015-01-08:
Here is the current HL2 dev state for HID Project 2.1:
https://github.com/NicoHood/HoodLoader2/tree/dev

keep in mind that the new 2.0.3 hasnt been added yet. Its also not really needed, just a typo
You just need to update your hardware folder with these files. it also only work with a 16u2 right now (should be the default)

bgma, 2015-01-09:
Now I have tried to compile the examples and they work to compile.
Some of the examples I have also tried to download into a "Arduino UNO R3 / Arduino-Compatible" not all but the majority of them.

Thank you very much!

NicoHood, 2015-01-09:
May you explain me what doesnt work exactly? I dont really understand what you mean by that.

bgma, 2015-01-09:
I meant that I tried to Compile / Verify all examples associated new HID 2.0.3 and it works fine. Some of the examples (not all) I uploaded to Uno and tested, and what I can confirm that they work.

So everything works fine.
/ BG

NicoHood, 2015-01-09:
So there is no problem anymore?
You are not able to compile a gamepad file if no gamepad core is selected though.

If everything is working fine, may you close the issue?

bgma, 2015-01-09:
Yes, as far as I can see, it works, except "AdvancedGamepad" which gives error at compile time. I must have missed it when I tested, sorry about that! has no gamepad.

closed 2015-01-09 by bgma

NicoHood, 2015-01-09:
Should be fixed now. Thanks again!