NicoHood/HID #9: Selected board depends on 'HID' core (not installed)

Snower4344, 2015-01-15:
Hi Sir,
I use your [Arduino HID Project 2.1] on the Leonardo board.
But I can not upload the program.
Please see the picture. Thank you very much

NicoHood, 2015-01-15:
Ensure that you rename the HID-master correct to HID. You can also see the note/pic in the readme:
https://github.com/NicoHood/HID/tree/Official-Integration#3-install-the-hid-core-files

Snower4344, 2015-01-16:
Hi sir,
But I can not find the folder[HID-master] in your HID-Official-Integration.zip file
I unzip the HID-Official-Integration.zip and copy all files to "C:\Program Files (x86)\Arduino\hardware\arduino" .

NicoHood, 2015-01-16:
And that is wrong. Read the readme again. You have to copy it into the sketchbook and not into your arduino folder.

closed 2015-01-25 by NicoHood

petersonsenha, 2015-06-11:
"Error 404, This is not the web page you are looking for"
https://github.com/NicoHood/HID/tree/Official-Integration#3-install-the-hid-core-files

Thanks.

petersonsenha, 2015-06-11:
where I find HID-Official-Integration.zip ? 
Thanks!

NicoHood, 2015-06-11:
This is outdated.
Look at the wiki how to install:
https://github.com/NicoHood/HID/wiki/Installation

petersonsenha, 2015-06-12:
Thanks Nico!