NicoHood/HID #12: Keyboard.release compilation error

pilotniq, 2015-02-11:
I have a program that I was using with the standard Arduino USB Keyboard library.

I just installed the latest NicoHood HID library.

I'm getting:

    pinballController2.cpp.o: In function `loop': 
    /Users/xyzzy/Downloads/PinballController2.ino:78: undefined reference to `Keyboard_::release(unsigned char)'
    /Users/xyzzy/Downloads/PinballController2.ino:105: undefined reference to `Keyboard_::release(unsigned char)'
    collect2: error: ld returned 1 exit status

I have not had time to troubleshoot this myself yet.

pilotniq, 2015-02-11:
Quick follow-up. I'm using a Pololu A-star 32U4 board.

If I select the Arduino Leonardo or Pololu A-star 32U4 in the board menu, then I don't get the `Keyboard_::release` errors, but I don't have access to the USB Core menu.

If I select the USB Project HID Core Arduino Leonardo HID-Project, then I get the above errors when I verify code, even though I have selected  Serial + HID Core.

pilotniq, 2015-02-11:
I fixed the problem by removing `inline` from the Keyboard.h header file in avr/cores/hid/USB-Core/
Probably the code for inline functions has to be in the header file for inline to work?

Fixed in my local copy.

NicoHood, 2015-02-11:
Good catch! Thank you!
May you please try the new master? If it works, please close the issue :)
https://github.com/NicoHood/HID/commit/64f2a0a66b39bfd979f2cf37d9dbe0e82d474524

closed 2015-02-16 by pilotniq