NicoHood/HID #13: Add support for Pololu A-Star 32u4

pilotniq, 2015-02-11:
I'm trying to add support for the Pololu A-Star 32u4 that I'm using. I assume it should be equivalent to the Leonardo.

I've copied the data from the boards.txt that came with the A-Star 32u4, and tried to integrate it into the HID boards.txt, by copying missing stuff from the Leonardo entry. I ended up with:

    aStar32U4.name=Pololu A-Star 32U4
    aStar32U4.vid=0x1ffb
    aStar32U4.pid=0x2300
    aStar32U4.vid.1=0x1ffb
    aStar32U4.pid.1=0xa300
    aStar32U4.upload.protocol=avr109
    aStar32U4.upload.maximum_size=28672
    aStar32U4.upload.speed=57600
    aStar32U4.upload.disable_flushing=true
    aStar32U4.bootloader.low_fuses=0xff
    aStar32U4.bootloader.high_fuses=0xd0
    aStar32U4.bootloader.extended_fuses=0xc8
    aStar32U4.bootloader.path=caterina
    aStar32U4.bootloader.file=Caterina-A-Star.hex
    aStar32U4.bootloader.unlock_bits=0x3F
    aStar32U4.bootloader.lock_bits=0x2F
    
    aStar32U4.build.mcu=atmega32u4
    aStar32U4.build.f_cpu=16000000L
    aStar32U4.build.core=HID:hid
    aStar32U4.build.vid=0x1ffb
    aStar32U4.build.pid=0x2300
    aStar32U4.build.usb_product="Pololu A-Star 32u4"
    aStar32U4.build.board=AVR_LEONARDO
    #aStar32U4.build.variant=arduino:leonardo
    aStar32u4.build.extra_flags={build.usb_flags}
    
    #USB core selection
    #HID Project needs to be installed https://github.com/NicoHood/HID
    aStar32U4.menu.usbcore.hid=Serial + Extended HID
    aStar32U4.menu.usbcore.hid.build.variant=leonardo_hid
    aStar32U4.menu.usbcore.gamepad=Serial + Gamepad HID
    aStar32U4.menu.usbcore.gamepad.build.variant=leonardo_gamepad
    aStar32U4.menu.usbcore.custom=Serial + Custom HID
    aStar32U4.menu.usbcore.custom.build.variant=leonardo_custom
    aStar32U4.menu.usbcore.USB_CORE=Default Core
    aStar32U4.menu.usbcore.USB_CORE.build.variant=leonardo
    aStar32U4.menu.usbcore.NO_USB=No USB functions
    aStar32U4.menu.usbcore.NO_USB.build.variant=leonardo_no_usb

When I try to build in the Arduino environment, I get the errors starting with:

    In file included from /Users/erlandlewin/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/CDC.h:51:0,
                     from /Users/erlandlewin/Documents/Arduino/hardware/HID/avr/cores/hid/Arduino.h:245,
                     from /Users/erlandlewin/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBAPI.h:53,
                     from /Users/erlandlewin/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:39:
    /Users/erlandlewin/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:101:87: error: 'USB_VID' was not declared in this scope
     D_DEVICE(USB_DEVICE_CLASS_IAD, USB_DEVICE_SUB_CLASS_IAD, USB_DEVICE_PROTOCOL_IAD, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);

I'm not sure how to continue from here. Can you help me? Thanks!

pilotniq, 2015-02-11:
I had success with a work-around of just adding the A-Star's vid and pid to the Leonardo entry.

Still wondering why the above doesn't work, though.

NicoHood, 2015-02-11:
So you added it to the library, and did not create your own? (because if you did, you have to provide a platform.txt file as well).

also the stuff youve added seems to be outdated. in 1.6 there is not bootloader.path vairable. its all in one entry now.

I suggest you to compare the leonardo entry again with the other settings. maybe copy it and only change the name, pid,vid etc. it should be very easy to do so.

Also i dont know which bootloader they use for their board. but it should work.

Also always add a link to the board/files you are talking about, so i have more infos:
https://github.com/pololu/a-star/blob/master/add-on/pololu/boards.txt

if you have a solution, feel free to post it here. I could add this to the wiki as well. If you still need more help, let me know

NicoHood, 2015-02-11:
I will add an example for the pro Micro tomorrow here:
https://github.com/NicoHood/HID/wiki/Adding-custom-boards.txt

Please let me know if the other issue you opened is now resolved and close it then.

NicoHood, 2015-02-17:
See this:
https://github.com/NicoHood/HID/wiki/Adding-custom-boards.txt#1-edit-existing-boardstxt-file

https://github.com/sparkfun/SF32u4_boards/pull/13

trevor403, 2015-02-23:
Your custom SF32u4_boards now makes HID 2.1 work but this issue should stay open because USB HID Gamepad mode controls the mouse ( you don't have issues enabled on your SF32u4_boards fork)

NicoHood, 2015-02-23:
First: You cannot open issues on forks and I dont see if someone copied it (at least I dont check every fork).

2nd: What? Can you describe this a bit better? You mean the gamepad axis controls the mouse?
What OS are you using? What board are you using?
I think think might happen if you change the descriptors with the same PID/VID. The new version (hoodloader2 only) will have a pid for each descriptor, so the OS recognizes the MCU as different device then. The problem is if th device is a keyboard today and a gamepad tomorrow the OS will be confused. Under Windows8 there is a simple fix:
https://github.com/NicoHood/HID/wiki/Troubleshoot-FAQ#switching-the-hid-core

May you give me more information? May you try this fix?

The reason why I cannot provide a seperate PID for Leonardo is simply because it requires new drivers. And not everybody wants to instal them. For HoodLoader2 you have to install them anyways so I dont care if I add one ore 3 more PIDs. Thats all. I could change the boards.txt to manually change the PIDs easier and use the Teensy driver instead. What do you think? Can you follow?

Did you also had a chance to test the new Gamecube library?

trevor403, 2015-02-24:
Yes I tested the new Nintendo library and it worked like a charm. I was going to load Hoodloader2 onto my Mega 2560 but the mega16u2 bootloader got corrupted while loading the self-install sketch. So i used a Pro Micro as an Arduino ISP to restore it. Technically I could use the same technique to install an already compiled Mega 2560 Hoodloader 2.0.3 hex but I don't want to do that right away. 

So I cloned your project to add Pro Micro support for IDE 1.6.0 and I installed the Gamecube_USB_HID sketch on the board with Gamepad as the USB core. I got it compiled and loaded onto the board but instead of registering as a joypad it registered as a Mouse, with the main stick being mouse x and y, and A being left click, and B being right click. The only other working button was moving right potentiometer to 256, that caused a left click (Im not sure if it was left or right potentiometer). The rest were dead. I am using Mac OS X 1.9.4 on my system. I know you disabled hoodloader support for 32u4, do you think it would help? Also I will try loading Hoodloader2 hex onto my Mega using ISP, see if that gives a natural joypad hid device. Thanks for reading and feel free to ask more questions if you need.

NicoHood, 2015-02-24:
Okay cool!

I did not disable HoodLoader2 support. I just did not add different PIDs and new drivers. Also the fix I am talking about is only in the new dev version I am currently working on. Make sure to update HoodLoader to 2.0.4 when its out, it fixes a usb hub problem.

But what you could do in the meantime is to change the Gamepad PID. I changed the boards.txt so its easier for you to change it. Download this file and change the PID so any other value:
https://github.com/NicoHood/HID/commit/8fb1a2cb0436c199a685a68e4d5887363f0e1b0f

Since there are no drivers required for Mac it should just work. On windows you need to create a driver file or install the Teensy drivers which work for all VIDs and PIDs.

NicoHood, 2015-02-27:
Working now? May I close this or does anyone need more information?

trevor403, 2015-02-28:
I'm running all the code now, ill write an update in like an hour, thanks again for the help.

NicoHood, 2015-02-28:
An update? where? here?

NicoHood, 2015-04-05:
May I close this issue?

trevor403, 2015-04-05:
Today running all tests on code fixes using same OS

closed 2015-04-05 by NicoHood