NicoHood/HID #14: Gamepad recognized as Mouse

NicoHood, 2015-02-27:
People reported me that the Gamepad axis is recognized as mouse.
I need more information to debug this.If you have the saame problem:

what OS are you using?
What board do you use? Micro/Leo or Mega/uno?

Under Windows 8 try this and tell me if it works for you:
https://github.com/NicoHood/HID/wiki/Troubleshoot-FAQ#switching-the-hid-core

trevor403, 2015-03-17:
OS - Mac OS X 10.9.4
Board - Pro Micro 16MHz 5v
USB Entry - Bus 006 Device 003: ID 2341:8037 2341 Arduino Micro

Description:
Gamecube_Test will show proper readings from the device. When using Gamecube_USB_HID sketch the mouse is moved to the top right corner of the screen. The A button acts as left mouse click. Right analog at position 256 also performs a left click. The rest I have not been able to test due to the mouse position.

NicoHood, 2015-03-17:
Please test the gamepad example of the HID Project itself.

However this seems to be wrong...
Maybe Mac doesnt like the mouse.

If you compile with the Micro outcomment line 32 and 34 here. If you compile with Leonardo do it for the equivalent leonardo file.
https://github.com/NicoHood/HID/blob/master/avr/variants/micro_gamepad/pins_arduino.h#L32-L34

Try again and replug the device. If its possible to remove the drivers under mac then try to do it.

trevor403, 2015-03-17:
I discovered that commenting out those lines makes the Gamepad HID work with no issue. But why don't you comment them out in master branch? If that header file is specifically for the micro, you should commit the change. Anyways the sketch is working just fine. I'm switching the dpad over to buttons, otherwise its all good!

NicoHood, 2015-03-17:
It seems that the MAC OS cannot handle a keyboard, mouse and gamepad together.

Maybe its just a small thing in the report descriptor that makes it incompatibel.

Can you please test to only disable the mouse or keyboard? To see if the mouse or the keyboard causes this problem or both together.

I also found a bit more information about force feedback, maybe I'll rework the gamepad anyways. But this will take some more time. I will possibly commit the change after you tested the keyboard/mouse only + gamepad thing i described above.

trevor403, 2015-03-17:
I tried to enable each of these with only `#define HID_GAMEPAD_ENABLE` and the results were all the same.
```
#define HID_MOUSE_ENABLE
#define HID_KEYBOARD_LEDS_ENABLE
#define HID_KEYBOARD_KEYS_ENABLE
```

In every instance, except for having `#define HID_GAMEPAD_ENABLE` as the only device enables, the mouse moves to the top right hand corner of the screen and stays there. It looks like it could be a usb descriptor problem to me.

NicoHood, 2015-03-17:
what if you only do Gamepad.begin() and Gamepad.Write() and then an empty loop? Does this also break the mouse?
Interesting. Seems that Mac doesnt like gamepads + other devices.

Have you ever tried HoodLoader1? Its quite old, just wondering if I made a new mistake or the Arduino IDE causes this problem.

trevor403, 2015-03-17:
I haven't tried HoodLoader1 you took 32u4 support out of HoodLoader2 recently so I never tried it on my Pro Micro's but with the empty loop setup you mentioned, and begin and write in setup(), I am able to have all three going at once without any mouse breakage.

```
#define HID_MOUSE_ENABLE 
#define HID_KEYBOARD_LEDS_ENABLE
#define HID_GAMEPAD_ENABLE
```

I think the issue may be in the way that Gamepad.buttons or something sends a USB call, though the PAGE ID should account for each device type sending a request.

NicoHood, 2015-03-17:
HL1 or HL2 was never intended to be used with a 32u4. it is a bootloader. The USB-Core is THIS HID-Project and the SAME. I never stopped support, I never had support, there is no such thing like support since it is something totally different. HL2 is a bootloader that enables to use this HID Project with the 16u2. HL1 was a special 16u2 firmware though.

Anyways:
Seems to be an OS issue. I have to look at the gamepad report itself but for now this should do it:
https://github.com/NicoHood/HID/tree/NicoHood-Gampad-Fix
Can you confirm this version works? Then I'll merge it the next days. Want to have a look to the report again first.

NicoHood, 2015-03-17:
Teensy also has a mouse + keyboard + joystick report. Do you have a chance to test a teensy?

Have you ever tried to change the PID in the board.txt?

NicoHood, 2015-04-05:
Gamepad fix merged into dev branch

closed 2015-04-05 by NicoHood