NicoHood/HID #15: Comiling issue on Pro Mirco boards

trevor403, 2015-02-28:
Have a compiling error goes:
```
Arduino: 1.6.0 (Linux), Board: "Arduino Micro HID-Project, Default Core"

Build options changed, rebuilding all
In file included from /home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/CDC.h:51:0,
                 from /home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/Arduino.h:245,
                 from /home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBAPI.h:53,
                 from /home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:39:
<command-line>:0:10: error: 'build' was not declared in this scope
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.h:327:60: note: in definition of macro 'D_DEVICE'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                            ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:101:96: note: in expansion of macro 'USB_PID'
 D_DEVICE(USB_DEVICE_CLASS_IAD, USB_DEVICE_SUB_CLASS_IAD, USB_DEVICE_PROTOCOL_IAD, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
                                                                                                ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.h:327:95: error: braces around scalar initializer for type 'u16 {aka short unsigned int}'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                                               ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:101:1: note: in expansion of macro 'D_DEVICE'
 D_DEVICE(USB_DEVICE_CLASS_IAD, USB_DEVICE_SUB_CLASS_IAD, USB_DEVICE_PROTOCOL_IAD, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
 ^
<command-line>:0:10: error: 'build' was not declared in this scope
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.h:327:60: note: in definition of macro 'D_DEVICE'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                            ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:115:93: note: in expansion of macro 'USB_PID'
 D_DEVICE(USB_DEVICE_NO_CLASS, USB_DEVICE_NO_SUB_CLASS, USB_DEVICE_NO_PROTOCOL, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
                                                                                             ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.h:327:95: error: braces around scalar initializer for type 'u16 {aka short unsigned int}'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                                               ^
/home/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:115:1: note: in expansion of macro 'D_DEVICE'
 D_DEVICE(USB_DEVICE_NO_CLASS, USB_DEVICE_NO_SUB_CLASS, USB_DEVICE_NO_PROTOCOL, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
 ^
Error compiling.
```

trevor403, 2015-02-28:
NOTE: using both dev branches

NicoHood, 2015-03-01:
Can you please post you boards.txt file?

trevor403, 2015-03-01:
Using the dev branch on the HID project and the SF32u4_boards project (patch-1)

Therefore here is my copy of boards.txt, https://github.com/NicoHood/SF32u4_boards/blob/patch-1/sparkfun/avr/boards.txt

trevor403, 2015-03-01:
I'll try using the master branches

NicoHood, 2015-03-01:
It compiles just fine for me. Just downloaded my fork and inserted to the hardware folder.

trevor403, 2015-03-01:
Yes the master branches compile just fine, the only issue was HID project's dev branch, you can close if you have no more questions.

NicoHood, 2015-03-01:
Does the new dev branch not compile now? I updated it a few hours ago and it just works fine. May you try it again and see if it works? If the bug I patched also caused this issue... then It is very weird.

trevor403, 2015-03-01:
Everything compiles alright with SF32u4_boards (regardless of master or patch-1) but only master works for HID project in conjunction, if i use dev branch this error pops up again (plus more),

```
/Users/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.h:327:95: error: braces around scalar initializer for type 'u16 {aka short unsigned int}'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                                               ^
/Users/trevor/Documents/Arduino/hardware/HID/avr/cores/hid/USB-Core/USBCore.cpp:115:1: note: in expansion of macro 'D_DEVICE'
 D_DEVICE(USB_DEVICE_NO_CLASS, USB_DEVICE_NO_SUB_CLASS, USB_DEVICE_NO_PROTOCOL, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
```

NicoHood, 2015-03-01:
What OS are you using?

What USB Core and what Board have you selected? (pro micro 5v??)

That is weird because it compiles under Windows. But there is also a bug related to this issue. I can imagine where this comes from, but I dont knwo why the dev version has this bug and the master doesnt.

trevor403, 2015-03-01:
I have tested with default and gamepad USB Cores on both Linux and Mac, So Unix systems seem to differ when it comes to usb.

NicoHood, 2015-03-01:
No. They differ when it comes to Arduino IDE compiling. I will test linux tomorrow hopefully.

trevor403, 2015-03-17:
The compiling error does not occur when using HID 2.1, using HID 2.2 causes the error.

NicoHood, 2015-03-17:
I cannot reproduce this. may you post more error messages or upload your sketchbook somewhere?
Had no chance to test linux still.

pierrotm777, 2015-03-30:
I have also this issue but only when i choose Arduino micro as board !
In file included from C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid/USB-Core/CDC.h:51:0,
                 from C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid/Arduino.h:245,
                 from C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBAPI.h:53,
                 from C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:39:
<command-line>:0:10: error: 'build' was not declared in this scope
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.h:327:60: note: in definition of macro 'D_DEVICE'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                            ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:101:96: note: in expansion of macro 'USB_PID'
 D_DEVICE(USB_DEVICE_CLASS_IAD, USB_DEVICE_SUB_CLASS_IAD, USB_DEVICE_PROTOCOL_IAD, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
                                                                                                ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.h:327:95: error: braces around scalar initializer for type 'u16 {aka short unsigned int}'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                                               ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:101:1: note: in expansion of macro 'D_DEVICE'
 D_DEVICE(USB_DEVICE_CLASS_IAD, USB_DEVICE_SUB_CLASS_IAD, USB_DEVICE_PROTOCOL_IAD, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
 ^
<command-line>:0:10: error: 'build' was not declared in this scope
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.h:327:60: note: in definition of macro 'D_DEVICE'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                            ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:115:93: note: in expansion of macro 'USB_PID'
 D_DEVICE(USB_DEVICE_NO_CLASS, USB_DEVICE_NO_SUB_CLASS, USB_DEVICE_NO_PROTOCOL, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
                                                                                             ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.h:327:95: error: braces around scalar initializer for type 'u16 {aka short unsigned int}'
  { 18, 1, 0x200, _class,_subClass,_proto,_packetSize0,_vid,_pid,_version,_im,_ip,_is,_configs }
                                                                                               ^
C:\tmp\Arduino Compilateur\arduino-1.6.1\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:115:1: note: in expansion of macro 'D_DEVICE'
 D_DEVICE(USB_DEVICE_NO_CLASS, USB_DEVICE_NO_SUB_CLASS, USB_DEVICE_NO_PROTOCOL, 64, USB_VID, USB_PID, 0x100, IMANUFACTURER, IPRODUCT, 0, 1);
 ^
Erreur lors de la compilation.

pierrotm777, 2015-03-30:
I confirme that i have this issue with the 2.2 version and not the 2.1 ...

NicoHood, 2015-03-30:
Thank you. I was to blind to check this. Now fixed:
https://github.com/NicoHood/HID/commit/f991009f73d639471bef6c7a7d4caed43180e03f

Does the Pro Micro work for you as well? So I can close the initial issue.

pierrotm777, 2015-03-30:
It's ok for me now with 1.6.1 and 1.6.2 ...

NicoHood, 2015-03-30:
I havent tried 1.6.2 at all (downloading now). Shouldnt cause too much problems I think.

Thx for the Feedback!

closed 2015-03-30 by NicoHood