NicoHood/HID #16: Compiling issue after commit #2289569

Ralim, 2015-03-01:
After latest commit, no longer compiles the HID_Consumer test code :/

Arduino: 1.6.0 (Windows 8), Board: "Arduino Leonardo HID-Project, Serial + Extended HID"

In file included from C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid/USB-Core/HID.h:435:0,
                 from C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid/Arduino.h:246,
                 from C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid\USB-Core\USBAPI.h:53,
                 from C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid\USB-Core\USBCore.cpp:39:
C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid/USB-Core/System.h: In member function 'void System_::press(uint8_t)':
C:\Users\Ralim\Documents\Arduino\hardware\HID\avr\cores\hid/USB-Core/System.h:64:4: error: 'USBDevice' was not declared in this scope
    USBDevice.wakeupHost();
    ^
Error compiling.

Also,
If i revert to the previous commit the media keys seem to no longer work?

NicoHood, 2015-03-01:
May you please use the current master? I will have a look into that soon. The dev versions can be really buggy. But thx for testing and reporting!

Ralim, 2015-03-01:
I tried the current master and couldn't get it to work either (thus testing dev, and reporting this)

NicoHood, 2015-03-01:
What board are you using, what OS and what do you mean by "does not work"?

Have you tried this?
https://github.com/NicoHood/HID/wiki/Troubleshoot-FAQ#switching-the-hid-core

Ralim, 2015-03-01:
Tried leanardo and micro pro boards.. Both are picked up as the keyboard under windows (8.1 x64) but do not register any of the media keys. Tried uninstalling drivers and reinstalling and the board is picked up with the keyboard but windows doesn't seem to pickup the keys..  

Is there any way for me to provide diagnostic information to help ?

NicoHood, 2015-03-01:
First: The issue is now fixed, thx!
https://github.com/NicoHood/HID/commit/e9748a7a8fcae7e711fef88eda9e3185a1cfcb8c

Yeah you can, and it should work. Go to devices and printers. The leonardo should appear there. Rightclick it and take a screenshot.

How did you uninstall the device? Please follow the instructions of the wiki page i linked above. Then unplug the device and uninstall it again. Then replug it and it should work. You can also join the chat, klick the link on the readme page.

Ralim, 2015-03-02:
Here is the right click menu : 
[image]
Would it be better for me to open a new issue?

NicoHood, 2015-03-02:
To make things more clear please open a new issue, yes.

You have to click on Arduino leonardo and then right click it. I think it was out of focus.

Are you using windows 8 or 8.1?

closed 2015-03-02 by Ralim