NicoHood/HID #17: HID Consumer keys not registering under windows 8.1

Ralim, 2015-03-02:
Hi,
Currently the issue is windows is acknowledging (or not recieving?) the media control keys.
Using windows 8.1, by programming the HID_Consumer.ino file as a provided example to an arduino Leanardo or a Micro. The device shows up in Devices & Printers as shown, but triggering the button does not register.

Tested button functions :
Volume up
Volume down
Volume Mute
Play/Pause

[image]

NicoHood, 2015-03-02:
So you selected the Leonardo as board. Did you ensure it is the one from the HID project?

What USB-Core did you select? It should only work with the extended HID. If you selected anything else it shouldnt even compile.

How does you sketch look like? the normal example provided with this library?

Have you tried to change the PID?

It seems that Windows doesnt recognize the new board. I had this problem with the official Uno PID. Maybe for some reason your OS blocks the new devices or something like that.

You may also chat with me here (might be offline for a few hours soon):
https://gitter.im/NicoHood/HID?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge

NicoHood, 2015-04-05:
The dev branch fixes this issue.

closed 2015-04-05 by NicoHood