NicoHood/HID #19: Add Joystick profile

pierrotm777, 2015-03-30:
I don't understand very well what is the difference between a gamepad and a joystick .
I have tried to add a Joystick profile to your 2.2 code but not certain (can't test actually) that it run well .
But i can compile the code ...

Hid.h
#ifndef HID_REPORT_JOYSTICK
#define HID_REPORT_JOYSTICK(report_id)  /* 32 buttons and a throttle - just in case the game doesn't recognise a joystick with no analog axis */ \
	0x05, 0x01,							/* USAGE_PAGE (Generic Desktop) */ \
	0x09, 0x04,							/* USAGE (Joystick) */ \
	0xa1, 0x01,							/* COLLECTION (Application) */ \
	0x85, report_id,					/*   	REPORT_ID */ \
	/* 32 Buttons */ \
	0x05, 0x09,							/*   	USAGE_PAGE (Button) */ \
	0x19, 0x01,							/*   	USAGE_MINIMUM (Button 1) */ \
	0x29, 0x20,							/*   	USAGE_MAXIMUM (Button 32) */ \
	0x15, 0x00,							/*   	LOGICAL_MINIMUM (0) */ \
	0x25, 0x01,							/*   	LOGICAL_MAXIMUM (1) */ \
	0x75, 0x01,							/*   	REPORT_SIZE (1) */ \
	0x95, 0x20,							/*   	REPORT_COUNT (32) */ \
	0x55, 0x00,			                /*   	UNIT_EXPONENT (0) */ \
    0x65, 0x00,			                /*   	UNIT (None) */ \
	0x81, 0x02,							/*   	INPUT (Data,Var,Abs) */ \
	/* 8 bit Throttle and Steering */ \
	0x05, 0x02,							/*   	USAGE_PAGE (Simulation Controls) */ \
	0x15, 0x00,							/*   	LOGICAL_MINIMUM (0) */ \
	0x26, 0xff, 0x00,					/*   	LOGICAL_MAXIMUM (255) */ \
	0xA1, 0x00,							/*   	COLLECTION (Physical) */ \
	0x09, 0xBB,							/*   		USAGE (Throttle) */ \
	0x09, 0xBA,							/*   		USAGE (Steering) */ \
	0x75, 0x08,							/*   		REPORT_SIZE (8) */ \
	0x95, 0x02,							/*   		REPORT_COUNT (2) */ \
	0x81, 0x02,							/*   		INPUT (Data,Var,Abs) */ \
	0xc0,								/*   	END_COLLECTION */ \
	/*   Two Hat switches */ \
	0x05, 0x01,							/*   	USAGE_PAGE (Generic Desktop) */ \
	0x09, 0x39,							/*   	USAGE (Hat switch) */ \
	0x15, 0x00,							/*   	LOGICAL_MINIMUM (0) */ \
	0x25, 0x07,							/*   	LOGICAL_MAXIMUM (7) */ \
	0x35, 0x00,							/*   	PHYSICAL_MINIMUM (0) */ \
	0x46, 0x3B, 0x01,					/*   	PHYSICAL_MAXIMUM (315) */ \
	0x65, 0x14,							/*   	UNIT (Eng Rot:Angular Pos) */ \
	0x75, 0x04,							/*   	REPORT_SIZE (4) */ \
	0x95, 0x01,							/*   	REPORT_COUNT (1) */ \
	0x81, 0x02,							/*   	INPUT (Data,Var,Abs) */ \
	0x09, 0x39,							/*   	USAGE (Hat switch) */ \
	0x15, 0x00,							/*   	LOGICAL_MINIMUM (0) */ \
	0x25, 0x07,							/*   	LOGICAL_MAXIMUM (7) */ \
	0x35, 0x00,							/*   	PHYSICAL_MINIMUM (0) */ \
	0x46, 0x3B, 0x01,					/*   	PHYSICAL_MAXIMUM (315) */ \
	0x65, 0x14,							/*   	UNIT (Eng Rot:Angular Pos) */ \
	0x75, 0x04,							/*   	REPORT_SIZE (4) */ \
	0x95, 0x01,							/*   	REPORT_COUNT (1) */ \
	0x81, 0x02,							/*   	INPUT (Data,Var,Abs) */ \
	0x15, 0x00,							/*   	LOGICAL_MINIMUM (0) */ \
	0x26, 0xff, 0x00,					/*   	LOGICAL_MAXIMUM (255) */ \
	0x75, 0x08,							/*   	REPORT_SIZE (8) */ \
	0x09, 0x01,							/*   	USAGE (Pointer) */ \
	0xA1, 0x00,							/*   	COLLECTION (Physical) */ \
	0x09, 0x30,							/*   		USAGE (x) */ \
	0x09, 0x31,							/*   		USAGE (y) */ \
	0x09, 0x32,							/*   		USAGE (z) */ \
	0x09, 0x33,							/*   		USAGE (rx) */ \
	0x09, 0x34,							/*   		USAGE (ry) */ \
	0x09, 0x35,							/*   		USAGE (rz) */ \
	0x95, 0x06,							/*   		REPORT_COUNT (2) */ \
	0x81, 0x02,							/*   		INPUT (Data,Var,Abs) */ \
	0xc0,								/*   	END_COLLECTION */ \
	0xc0								/*   END_COLLECTION */
#endif

NicoHood, 2015-03-30:
A Joystick is nearly the same as the Gamepad. However the Gamepad is just USB Joystick Data. I had Problems with the "real" Gamepad. Its just a name.

What do you want to do? You can modify the report of the Gamepad (which is still Joystick data actually) and remove/add buttons. However you will always get problems If you increase the button or axis numbers. Also 2 Gamepads dont work well. People also had problems Under Mac with Mouse + Gamepad. (issue #14 )

I should provide a custom Gamepad example, I know. But I dont have time for it to code a 2nd API and report descriptor right now. As you can see development stuck a bit for the last weeks.

pierrotm777, 2015-03-30:
This is my code . My goal is to command a RC simulator with the arduino
micro .

NicoHood, 2015-03-30:
Just use the gamepad API?

pierrotm777, 2015-03-30:
Ok , i shall try the gamepad profile !

NicoHood, 2015-03-30:
Yeah. Its just a different name. It might be "too big" but simply ignore the other buttons/axis. Should be fine.

Close the issue if the gamepad works fine for you. Also see this fix if you have problems with it:
https://github.com/NicoHood/HID/tree/NicoHood-Gampad-Fix

NicoHood, 2015-04-05:
Any progress?

closed 2015-04-05 by pierrotm777