NicoHood/HID #35: Add Boards Manager install support

NicoHood, 2015-05-09:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

per1234, 2015-05-26:
I'm happy to add Boards Manager install support for this project and your HoodLoader2 bootloader and submit them via pull request. First you need to decide on where you want  the install files to be located. It's worth giving the location of the json file some thought because it's best not to change the URL after people have already added it to their preferences. So far these are the locations I have seen others use on Github, you could use a combination of these locations, one for the json and the other for the install file:

1. gh-pages branch. This is how [Adafruit](https://adafruit.github.io/arduino-board-index/package_adafruit_index.json) has theirs set up.
  - Pros: URL is shorter(NicoHood.github.io/HID/package_NicoHood_HID_index.json).
  - Cons: Adds another branch.
  - I would need you to create the gh-pages branch.

2. preexisting branch. This is the way [Sparkfun](https://github.com/sparkfun/Arduino_Boards/tree/master/IDE_Board_Manager) did theirs.
  - Pros: Doesn't add any more branches to the repository.  
  - Cons: Adds files that are useless to someone doing a manual install.
  - I would need you to decide where in the repository folder structure you want the files located.

3. Github releases. This is how [Sanguino](https://github.com/Lauszus/Sanguino/) has it set up.
  - Pros: You are already using Releases for your versions.
  - Cons: 
    - The folder structure of the Boards Manager install file is different than the manual installation file(no AVR folder) so this requires a separate Boards Manager install file to be added to each release. This may be confusing to users trying to figure out which file they need to download for manual installation but you could add wording to explain the purpose of the file.    
    - I don't think there is any way for releases to be added or modified via pull request so updates will always be your(or a project collaborator's) responsibility.
    - The json file could be added to a release but this is a bit confusing because a single json file contains all the versions and you don't want to change the URL with each version.
  - I would need you to add the install files to the releases.

Note: We used a zip file automatically created by Github for the [attiny](https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json) install file but Github later modified the file and the checksum changed, breaking the json file so I don't think that is an option.

PS. Sorry I provided you with a link to wrong information on this process when you had asked on the attiny pull request. You called them libraries and so I didn't realize what you were talking about.

NicoHood, 2015-05-27:
I am developing the files, so I need to keep the current structure.
So I think the avr folder needs to stay. Not sure if that gives any problems with the boards manager because I then have a manual install (with .git folder) and its also listed in the manager. But I guess its not a problem.

Also you have to remember that a lot of people dont have directly internet access. So you need to manually install it. Thatswhy I created an issue to install via zip file those boards, but they wont add it. Ergo I want to keep the structure for sure. As I can see option3 should be the best one.

The json can be located in the master branch, just in the default location so I can easy update it.
The installing code have to be added as you said. I have no idea how.

It would be so great if you can add a file via PR for HID and HL2. Please commit to the master, not the dev branch.

closed 2015-06-01 by NicoHood