NicoHood/HID #37: ‘class Keyboard_’ has no member named ‘writeKeycode’

pandacriss, 2015-05-22:
when I try to compile keyboard_keycode.ino example I get the following error message:
Keyboard_Keycode.ino:36:14: error: ‘class Keyboard_’ has no member named ‘writeKeycode’

although the keyboard.h have the implementation: size_t writeKeycode(uint8_t k); and keybord.cpp 
size_t Keyboard_::writeKeycode(uint8_t c) {...}

the other examples compile and run properly.
the same error occurs with arduino IDE 1:1.0.5 and 1.6.4 / 32 and 64 bits Linux mint 17.

pandacriss, 2015-05-22:
same error with windows XP. it show a longer error message:

Arduino:1.6.4 (Windows XP), Placa:"Arduino Leonardo"

Opciones de compilación cambiadas, reconstruyendo todo
Keyboard_Keycode.ino: In function 'void loop()':
Keyboard_Keycode:36: error: 'class Keyboard_' has no member named 'writeKeycode'
'class Keyboard_' has no member named 'writeKeycode'

 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:155)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:99)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:271)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:247)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:229)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:151)
	... 1 more
Caused by: java.net.UnknownHostException: downloads.arduino.cc
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1168)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1104)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:998)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:932)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:158)
	at cc.arduino.contributions.packages.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:97)
	... 5 more

  This report would have more information with
  "Show verbose output during compilation"
  activala desde Archivo > Preferencias
________________________________________________________________________________
in verbose mode:
________________________________________________________________________________
Arduino:1.6.4 (Windows XP), Placa:"Arduino Leonardo"

Opciones de compilación cambiadas, reconstruyendo todo

C:\Archivos de programa\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Archivos de programa\Arduino\hardware\arduino\avr\cores\arduino -IC:\Archivos de programa\Arduino\hardware\arduino\avr\variants\leonardo C:\DOCUME~1\panda\CONFIG~1\Temp\build386574210883240721.tmp\Keyboard_Keycode.cpp -o C:\DOCUME~1\panda\CONFIG~1\Temp\build386574210883240721.tmp\Keyboard_Keycode.cpp.o 
Keyboard_Keycode.ino: In function 'void loop()':
Keyboard_Keycode.ino:36:14: error: 'class Keyboard_' has no member named 'writeKeycode'
'class Keyboard_' has no member named 'writeKeycode'

RingoM, 2015-05-23:
According to your verbose log, I think you selected the regular leonardo board.
Please make sure you install the HID core files: [wiki](https://github.com/NicoHood/HID/wiki/Installation#3-install-the-hid-core-files)

and select the correct board (Arduino leonardo HID project): [wiki](https://github.com/NicoHood/HID/wiki/How-to-use-%28Micro-Leonardo-&-HoodLoader2%29)

pandacriss, 2015-05-23:
thank you. I did not notice that options. I should read all the documentation first.
the example seem that simple I did not notice.
should I delete the issue or leave it here so other users can find it?
muchas gracias.

NicoHood, 2015-05-23:
Thanks RingoM. I think I'd have searched for hours to find his bug :D 
Sorry, that I still had no time to review your PR. Just no time for anything.

I will close the issue, since its just a RTFM error ;)

closed 2015-05-23 by NicoHood