NicoHood/HID #40: Board manager with IDE 1.6.4 warnings

pierrotm777, 2015-06-03:
The board manager return me some warnings !
Warning: platform.txt from core 'HID Project USB-Core' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'HID Project USB-Core' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'HID Project USB-Core' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.

NicoHood, 2015-06-04:
Fixed in the dev branch already.

Thanks for the report anyways ;)

NicoHood, 2015-08-09:
IDE 1.6.6 can be tested now. Those setting dont matter then at all.
https://github.com/NicoHood/HID/tree/dev_2_4

closed 2015-08-09 by NicoHood