NicoHood/HID #46: USB wake up not working

raaj5671, 2015-08-06:
I am trying to wake up my PC using my arduino leonardo, but currently I can open the power management as a keyboard in my device manager. I also have enabled to to wake my computer but now I still cannot wake my computer when I transmit data to my computer through arduino. Can i know why. I am kinda newbie to arduino. Thank you.

NicoHood, 2015-08-06:
There is an example that shows you how to do that. Look at the Consumer keys example.

raaj5671, 2015-08-06:
how do i get there?

NicoHood, 2015-08-06:
Oh it was System. Its all included in the .zip file via examples.
https://github.com/NicoHood/HID/blob/master/examples/HID/System_Basic/System_Basic.ino

closed 2015-08-06 by NicoHood

bam80, 2015-09-23:
Thank you a lot!
Maybe emphasize it in FAQ of the HID project, or elsewhere?
I wandered about this feature also, but just didn't know where to search/ask..

NicoHood, 2015-09-23:
This feature has been added to the Arduino Core now and will be published with 1.6.6 to the upstream.

bam80, 2015-09-23:
Will it wake up PC from S3 suspend state?
When PC goes to sleep by itself, will it inform USB devises about the state change, and if so - will Arduino handle such report somehow?
Sorry about asking here, just don't know where to ask elsewhere

NicoHood, 2015-09-23:
Yes. But you need to activate usb wakeup in the bios. Some laptops have this disabled to save power.
Just try it. with the older or newer hid project, doesnt matter. the wiki is updated to the 2.0.4 dev version. I know there is documentation to be improved.

bam80, 2015-10-06:
Thank you NicoHood,
could you answer to the second half of my previous question also?
As I understand, just before sleeping, host reports to all it's USB devices that it going to sleep, so that they always knew the actual state of host. The same occurs when host wakes up. I might be wrong, but it's my guess.
If I am right, does that functionality implemented?
If not, how would I know if the host is sleeping, and I need to call System.write(SYSTEM_WAKE_UP)
to wake it up before sending anything to it?

NicoHood, 2015-10-06:
I dont know. Look at USBCore.cpp in the arduino core (master, nightly). This is how they implemented it. You may want to have a look at the 32u4 datasheet and the usb specs.

I did not implement, nor invent this myself. So you could only ask the guy who originally created the arduino PR. https://github.com/arduino/Arduino/pull/1488/files?diff=split

Sorry for no better answer.

bam80, 2015-10-06:
Ok, if things are going to be such complicated,
could this be workaround for a while:
since we don't know if the host is sleeping or not,
can we just always send System.write(SYSTEM_WAKE_UP) along with any actual data we are going to send?
Don't you see any difficulties with this approach? I'm going to implement IR remote controller, if that matter

NicoHood, 2015-10-06:
Try it out. Let me know if you have anything new. You could still ask in the PR I linked. There are some very clever people listening there that you can ask.

bam80, 2015-10-06:
Thank you for the direction,
I just realized that I don't need to wake up the system on every key on the remote - doing so on just Power key is enough. So that tricky logic for tracking host sleep states just not needed here, sorry for disturb you)

bam80, 2015-10-08:
Hi Nico, thank you again for the PR link - it exactly what I was searching for.
Just apologize it seems the is lack of documentation on this, and things happens confused,
in particular, it was not obvious for me (and written nowhere), that
*System.write(SYSTEM_WAKE_UP)*
and
*Keyboard.systemControl(SYSTEM_CONTROL_WAKEUP)*
are not the same,
and the former will wake PC from S3 suspend,
but the latter rather not.
I figured this out only by looking into the source code

NicoHood, 2015-10-08:
They should be the same. They should both call USBDevice.wakeup() (or similar syntax).
The wakeup however is not part of the System HID device. The System HID Device is only important to actually shut down the PC, but not to wake it up.

Bootkeyboard etc is all coming. However the dev branch is broken for now, but a lot of work is going on. Not sure if you noticed it: https://github.com/arduino/Arduino/pull/3931 And many more discussion before. This is getting great!

If you are currently working with USB and it looks like you are more experienced, you may want to test the dev branch once its fixed for the new hourly IDE. I think I will do this the weekend.

bam80, 2015-10-08:
According to [this post](http://forum.arduino.cc/index.php?topic=150157.msg2427365#msg2427365), they are not:
*Keyboard.systemControl(SYSTEM_CONTROL_WAKEUP)* doesn't call USBDevice.wakeup() and thus is not able to wake from S3 suspend,
whereas *System.write(SYSTEM_WAKE_UP)* does.

I doubt I am more experienced with USB, my initial intention was to create IR adaptor for my remote, which happens to be IR surprisingly. So I am just putting the pieces together.

However, I don't mind to learn by the way.
It's great the things getting better.
I would be glad to test it with the new IDE when it will be ready.

NicoHood, 2015-10-08:
Oh I didnt know. Yes the System class cannot wakeup the PC. The variable is just a wrapper to USBDevice.wakeup() if you look at the source.

I am doing the same actually. Just that I also develop my own libraries. An example on how to setup a nec remote to press HID Keys will follow (In the IRLremote I think). Or I will open a new repository with all my personal project sketches. It was working amazing. But now the libraries need an update, but the time will come when everything is ready ;)

bam80, 2015-10-08:
Very interesting to see your sketches. I saw some third parties but wonder if yours are different