NicoHood/HID #53: Support for rumble/force feedback

AGausmann, 2015-08-29:
*Clarification of #52*

There seems to be a missing feature in the Gamepad class. I would like to be able to handle feedback from the host controller/program, for example how the rumble or force feedback feature is provided by some gamepads. I believe some related code is in place (such as keyboard LEDs) that just has to be expanded upon to include this feature.

NicoHood, 2015-08-30:
FB is a very complicated thing. Only one person has been able to get this working yet. Its a N64/Gamepcube Adapter. I think it was blissbox. FB needs a lot more complicated stuff to implement, I never got it running. If you have any documents on how to implement, feel free to post them. But for now I have no idea how to do this. Also I dont have the PC side test environment of supported games.

You can still send data via the CDC serial to the host, if it does not have to be a real FB rumble function of the gamepad. Use this instead. Maybe RAW HID could also be an option, once its implemented (not done yet).

AGausmann, 2015-08-30:
My primary research was in the HID class specs (http://www.usb.org/developers/hidpage/HID1_11.pdf) however I have very little experience in this topic. From what I see in Appendix B and D (example descriptors), it may be as simple as putting an Output in the device descriptor.

NicoHood, 2015-08-30:
Well the HID library is pluggable. You can try it yourself. You can modify my code or start from a new gamepad. Try it, I wont do this right now, as said I cannot even test this, I dont own games that support rumble.

Dont get me wrong, I really want to see this feature. But I dont think its possible and I dont have the time to look into this minor feature. I try to keep the main features alive and port the library to 1.6.6. the Keyboard out report is still missing, I also dont know how to do this yet. So I also dont have an idea how to do this for Gamepad. If I have more infos, I let you know, but dont expect me to come up with a rumble function myself.

AGausmann, 2015-08-30:
Okay, I'll do a bit of research and testing to see what needs to be done.

NicoHood, 2015-08-30:
http://www.bliss-box.net/Bliss-Box/index.html
http://www.raphnet.net/electronique/gc_n64_usb/index_en.php

The last link has real force feedback with a HUGE HID Descriptor. Not sure what else have to be changed. Maybe this helps you, thats the only useful and working implementation that I found in all my research.

AGausmann, 2015-08-30:
Would it be correct to branch off `dev_2_4`? Currently, that branch fails to compile (missing HID.h)

NicoHood, 2015-08-30:
this branch is correct.

https://github.com/NicoHood/HID/wiki/Installation

AGausmann, 2015-08-30:
It's missing HID.h which is included by most of the other files.

NicoHood, 2015-08-31:
Have you used the IDE from the PR?
I think this is related to this issue:
https://github.com/arduino/Arduino/issues/3743

Please delete the core in appdata/.arduino15 if you have installed it.

AGausmann, 2015-08-31:
Yep, I forgot to update to 1.6.6... Nice catch, I'll update the core when I upgrade

NicoHood, 2015-09-01:
You should NOT upgrade the core X_x Only the IDE.

timemaster5, 2019-04-17:
This project seems to has a force feedback support:
https://github.com/tloimu/adapt-ffb-joy

This commit seems to add the base functionality:
https://github.com/tloimu/adapt-ffb-joy/commit/e4e5cd395b97def0a3d0ebcf10b331408d01c2a9

Nixon-Joseph, 2024-02-14:
Hey guys, sorry to resurrect such an old thread, but my bro and I are designing our own game controller, and this seems the best Libra for it. I've got 1 pin left on my microcontroller, and was hoping to use it for rumble.

Has anyone on this thread successfully integrated the rumble feature? I'll look into it when I have time if nobody has had any success. I figure it's been long enough that perhaps someone has.

@agausmann, @NicoHood, @timemaster5

agausmann, 2024-02-14:
Sorry I can't be of much help, I haven't touched this in literally 8 years