NicoHood/HID #55: Is this library only compatible with AVR (that's is to say: doesn't it work with Due nor Zero)?

q2dg, 2015-09-28:
I ask this because I only see a "avr" folder on github's root...and Arduino docs say (https://www.arduino.cc/en/Reference/Libraries) it's Keyboard and Mouse libraries are compatible with Due and Zero...
Thanks.

NicoHood, 2015-09-29:
But thats not an official library. And also the dev version is the one that COULD be compatible with due and zero later. But if they dont change the HID API I am not able to port it. Because the current HID implementation is not flexible enough and the Pluggable USB as well.

The master is not under development anymore, and it also only for AVR.

closed 2015-09-29 by NicoHood

q2dg, 2015-09-29:
Ok, thanks. Hoping to see if your HID library is merged into HID official's one....let's have faith