NicoHood/HID #58: BootKeyboard doesn't work with BootMouse

bam80, 2015-10-12:
If BootKeyboard and BootMouse used together in one sketch, only BootMouse works

NicoHood, 2015-10-12:
Hm.
Will check that later.

What if you combine a Bootkeyboard with Gamepad1 for example?

NicoHood, 2015-10-12:
Are you using a 16u2 or 32u4? If its a 16u2 you should read the wiki page API Documentation once again ;)

bam80, 2015-10-12:
Oh I see now, thank you:
>you can add up to 3 devices for the 32u4 and 1 for the 8/16/32u2

Is there any PR already to deactivating the CDC Serial?
I can't enter UEFI setup without it to check mouse functionality )

NicoHood, 2015-10-12:
I dont understand your 2nd sentence

With the 16u2 it is expected to not work. As written above please read the new wiki entry "API Documentation"

closed 2015-10-12 by NicoHood

bam80, 2015-10-12:
As I understand the new Documentation,
both BootKeyboard and BootMouse don't work with 16u2 because "I can add up to 1 device for the 8/16/32u2" but I need two devices to add.
Thus, if I free the device used by CDC Serial, I would get more devices available and could use both BootKeyboard and BootMouse, could I?

NicoHood, 2015-10-12:
Ah I see. Yes you are right. You could use a 32u4 instead.

Deactivating the Serial is not implemented and I dont know if the guys will ever support that. (since the reprogramming then needs a manual reset button press which is way too complicated for beginners, so we leave the serial always inside the core even if its not needed...). I will try to provide a patch for that if they dont do that. Could be a bit complicated though.

Pololu Mini LV is my favorite 32u4 board atm ;)

bam80, 2015-10-12:
Thank you,
the mouse is not really necessary for me in UEFI, I just wanted to test it because my UEFI setup apparently support it. So maybe it's not worth to spend time on it if it's too complicated.
Pololu seems too pricey for just a IR receiver.
What I look at is a [Teensy LC, $11.65](https://www.pjrc.com/store/teensylc.html), but I wonder if IRLremote could work on it?

NicoHood, 2015-10-12:
Nope ARM is not supported for IRLremote.
As alternative Genuino Micro board is now way more cheap than before. (also 12€ like the LC).
I like the pololu board since it supports 2.7V-11V which I can use with any power supply, especially I can just hook it up to my Led PSU.

UEFI would be nice if you can test it. Why not simple test BootMouse without keyboard? You can use a normal keyboard to enter the setup?

bam80, 2015-10-12:
Well, I just hoped because it seems like IRremote supports ARM)

As I said I have only bluetooth keyboard connected which is not boot USB capable.
I could find casual USB keyboard though.. Actually I will certainly find it if you decide to not spent time on deactivating the Serial ;)

NicoHood, 2015-10-12:
Just checked, the uefi that my dad has here doesnt set the protocol. Meaning it support normal HID devices. You might try to enter the setup with a normal keyboard. Then a normal mouse might also work.

I have some old pcs with mouse support at bios (well you can wiggle the mouse at startup and thats it) but this can be used to test it. Need to get my hands on that. Also I found the "bug":
https://github.com/arduino/Arduino/pull/3966

Linux and Windows just dont set the protocol at reenumeration. who cares about usb protocols?

bam80, 2015-10-13:
My UEFI support USB boot protocol only (Keyboard class doesn't work), so I guess it's suitable case for mouse testing.
Are you going to try deactivate the Serial near days?

NicoHood, 2015-10-13:
I think there is other work to do first. So probably now. But I may come up with a solution today or within month, you never know ;)
But a normal keyboard isnt so hard to find i guess...