NicoHood/HID #59: dev_2_4 break press/releaseAll at #1a676a2

amoriello, 2015-10-20:
Hi Nico,

I just synchronized my local fork from dev_2_4 fbbf4f0   to   dev_2_4 1a676a2  and noticed that the Keyboard.press/Keyboard.releaseAll() no longer works.

I used these to simulate "ENTER" e.g:

     Keyboard.press(0xB0);    //0xB0 : KEY_RETURN
     Keyboard.releaseAll();

Note that:

     Keyboard.print("yeah");

Still works as expected.

Thanks!

NicoHood, 2015-10-21:
because 0xB0 will trigger the ascii lookup table and not the key. I changed that and used enums.

All you need to do is:
Keyboard.press(KEY_RETURN);

A full list of all key will be added to the wiki soon. NKRO  keyboard will be updated as well. New features such as media key support inside the keyboard will be added too.

If that works for you, please close the issue. Thx for testing!

amoriello, 2015-10-21:
Works as expected, thanks.

Maybe should you consider indicating that Arduino define values described here (https://www.arduino.cc/en/Reference/KeyboardModifiers) no longer match in this library.

I'm following you on this choice.

Great work, thanks. :+1:

closed 2015-10-21 by amoriello

NicoHood, 2015-10-22:
Oh well the names DO work, but not the numbers. I will add the enum list to the wiki soon (I will add more buttons soon, so I will wait some time)