NicoHood/HID #60: Compilation termined

LuisDiazUgena, 2015-10-21:
Hi!

I've downloaded the dev 2-4 branch and I'm trying to test the library to make a keypad under arduino 1.6.6 (hourly build of today) and when trying to compile it says:

In file included from /home/luis-diaz/arduino-nightly/libraries/HID/src/HID-Project.h:41:0,
                 from /home/luis-diaz/arduino-nightly/libraries/HID/examples/Consumer/Consumer.ino:12:
/home/luis-diaz/arduino-nightly/libraries/HID/src/MultiReport/AbsoluteMouse.h:29:17: fatal error: HID.h: No such file or directory
 #include "HID.h"
                 ^
compilation terminated.
exit status 1
Error compiling.

I don't know what to do to fix this error. Would you help me?

Thanks!

NicoHood, 2015-10-21:
Check the examples in the branch. You need to include HID-Project.h instead.

Close if this solved the problem.

LuisDiazUgena, 2015-10-21:
I'm using the examples in the branch. This is what I'm doing:

#include "HID-Project.h"

const int pinLed = LED_BUILTIN;
const int pinButton = 2;

void setup() {
  pinMode(pinLed, OUTPUT);
  pinMode(pinButton, INPUT_PULLUP);

  // Sends a clean report to the host. This is important on any Arduino type.
  Consumer.begin();
}

void loop() {
  if (!digitalRead(pinButton)) {
    digitalWrite(pinLed, HIGH);

    // See HID Project documentation for more Consumer keys
    Consumer.write(MEDIA_PLAY_PAUSE);

    // Simple debounce
    delay(300);
    digitalWrite(pinLed, LOW);
  }
}

And the error is the same.

NicoHood, 2015-10-21:
Oh. Seems that it cannot find the built in library. What board are you using?

LuisDiazUgena, 2015-10-21:
I'm using a Leonardo clone from china

NicoHood, 2015-10-21:
Can you please post the full verbose compilation output?

NicoHood, 2015-10-21:
Tested your sketch with the latest 1.6.6 build (compiled on my own) and my dev 2.4 branch (with some not synced modifcations that should not matter) and it works.

LuisDiazUgena, 2015-10-21:
Arduino: 1.6.6 Hourly Build 2015/10/21 04:26 (Linux), Board: "Arduino Leonardo"

/home/luis-diaz/arduino-nightly/arduino-builder -dump-prefs -logger=machine -hardware "/home/luis-diaz/arduino-nightly/hardware" -hardware "/home/luis-diaz/.arduino15/packages" -tools "/home/luis-diaz/arduino-nightly/tools-builder" -tools "/home/luis-diaz/arduino-nightly/hardware/tools/avr" -tools "/home/luis-diaz/.arduino15/packages" -libraries "/home/luis-diaz/Escritorio/BQ/git/libraries" -libraries "/home/luis-diaz/arduino-nightly/libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10606 -build-path "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/examples/Consumer/Consumer.ino"
/home/luis-diaz/arduino-nightly/arduino-builder -compile -logger=machine -hardware "/home/luis-diaz/arduino-nightly/hardware" -hardware "/home/luis-diaz/.arduino15/packages" -tools "/home/luis-diaz/arduino-nightly/tools-builder" -tools "/home/luis-diaz/arduino-nightly/hardware/tools/avr" -tools "/home/luis-diaz/.arduino15/packages" -libraries "/home/luis-diaz/Escritorio/BQ/git/libraries" -libraries "/home/luis-diaz/arduino-nightly/libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10606 -build-path "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/examples/Consumer/Consumer.ino"
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library Snooze is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
"/home/luis-diaz/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"'  "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp/sketch/Consumer.ino.cpp"
"/home/luis-diaz/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/cores/arduino" "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/variants/leonardo" "-I/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src" "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp/sketch/Consumer.ino.cpp"
"/home/luis-diaz/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/cores/arduino" "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/variants/leonardo" "-I/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src" "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp/sketch/Consumer.ino.cpp"
"/home/luis-diaz/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/cores/arduino" "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/variants/leonardo" "-I/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src" "/tmp/builddb5c7c97be0cf0cd267221336a637cfa.tmp/sketch/Consumer.ino.cpp"
"/home/luis-diaz/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/cores/arduino" "-I/home/luis-diaz/.arduino15/packages/arduino/hardware/avr/1.6.8/variants/leonardo" "-I/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src" "/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src/HID-APIs/PluggableUSB.cpp"
[... cut ...]
                 from /home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src/SingleReport/BootKeyboard.h:31,
                 from /home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src/HID-Project.h:55,
                 from /home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/examples/Consumer/Consumer.ino:12:
/home/luis-diaz/arduino-nightly/libraries/HID-dev_2_4/src/SingleReport/../HID-APIs/ImprovedKeylayouts.h:47:1: error: expected declaration before '}' token
 };
 ^
exit status 1
Error compiling.

NicoHood, 2015-10-21:
Have you removed the old HID Project (hardware folder)?

NicoHood, 2015-10-21:
Oh and you placed it in the wrong folder. not nightly/libraries, you need to place it inside the sketchbook folder.

LuisDiazUgena, 2015-10-21:
thanks, will test it tomorrow. I've all on my work computer.

LuisDiazUgena, 2015-10-22:
Hi!

I've tested what you told me to do and it still doesn't work. I've checked ./arduino15 and removed HID-Project folder, then I've downloaded HID-dev_2_4 to mi sketchbook folder (~git/libraries) and it says exit status 1 (as before). (Here is the verbose output)[http://pastebin.com/4ApLFJmt]

NicoHood, 2015-10-22:
Weird. you could try to download a new Arduino nightly build. I used my own compilation.
So you are using linux? Which distribution?

-> I think i found the error. I complained about this some time ago.
https://github.com/arduino/Arduino/issues/3743

LuisDiazUgena, 2015-10-22:
Will try to download it again, but I've downloaded yesterday and it says the las update was on oct 14, so I think there is no point about this, but.. will try.

I'm using Ubuntu 14.04 LTS.

LuisDiazUgena, 2015-10-22:
I think there was something wrong about my computer, because I've donwloaded again and it works... Thanks for all your help!

NicoHood, 2015-10-22:
Weird. And you did not have to delete the avr core folder?

Feel free to close this issue and reopen it at any time when this error comes back.

LuisDiazUgena, 2015-10-22:
I do nothing else than download the IDE again and move the library into sketchbook folder. I'm having some issues with my computer and I'm going to make a fresh re-install of ubuntu 14.04, because I'm having some issues with other programs too.

Thanks again for all your help! I'm going to start playing right now!!

LuisDiazUgena, 2015-10-22:
In case someone needes, I just figured out that the Arduino IDE (1.6.6) needs to be runned as root to compile the sketch.

NicoHood, 2015-10-22:
This is not true. Why would you need this? Do you get this error still if you dont run it as root? It is possible, because root is another profile and this profile doesnt have the avr core which i was talking about. Please check that once again.

**This must work without root privileges, at least compiling**

LuisDiazUgena, 2015-10-22:
I've checked without launching arduino as root and it says (no verbose output):

In file included from /home/luis-diaz/Escritorio/BQ/git/libraries/HID-dev_2_4/src/HID-Project.h:41:0,
                 from /home/luis-diaz/Escritorio/BQ/git/Sound-Control/Leonardo-Sketch/Leonardo-Sketch.ino:1:
/home/luis-diaz/Escritorio/BQ/git/libraries/HID-dev_2_4/src/MultiReport/AbsoluteMouse.h:28:26: fatal error: PluggableUSB.h: No such file or directory
 #include "PluggableUSB.h"
                          ^
compilation terminated.
exit status 1
Error compiling.

NicoHood, 2015-10-22:
Could you please read my posts above again carefully and go to the other issue I linked. You need to delete the avr core inside your home folder (which differs from your roots home folder and this causes the problem to disappear).

LuisDiazUgena, 2015-10-23:
I've deleted every HID folder out of my sketchbook/libraries folder and run arduino as non-root user, but the problem is still there. 

Will check out in my home computer after work and see what's happens there. This computer is loaded with tons of programs and stuff, so I'm not sure if anything is messing around with this.

NicoHood, 2015-10-23:
You do not read carefully.

You need to delete the **avr core** inside /home/name/.arduino15/hardware/avrcore. Path might be similar, I dont have this core, so I cannot see it. This path has nothing to do with a) the sketchbook, b) a library and c) the HID Project.

This is an internal avrcore update for 1.6.5 but 1.6.6 still sees this as "latest and greatest" avr core and then missed the new pluggable usb etc. And your root user doesnt have those files because you never started 1.6.5 with root and updated the avr core. Thats what I think happened.

NicoHood, 2015-10-23:
what is inside ./arduino15?
Its package or whatever. Shouldnt contain much data, so please just look what you find in there. Shouldnt be so hard to find an avr core.

LuisDiazUgena, 2015-11-04:
Hi! I've moved from Ubuntu to Windows (due to work reasons) so I'm not going to be able to test this anymore, but I've installed it on Windows 7 Professional and it works with Arduino nightly 1.6.6.

Thanks for all your support.

NicoHood, 2015-11-06:
If you get any chance to debug this or have a solution, let me know.
Closing this for now.

A Proper Library Manager Update will follow. If you (or someone else) have the time to test, let me know the results.

closed 2015-11-06 by NicoHood

numist, 2016-06-19:
I'm running into the same error on Arduino 1.6.9, but using a Teensy2 board so I'm already pretty far off the reservation :/

```
"/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -fno-exceptions -felide-constructors -std=c++0x -mmcu=atmega32u4 -DTEENSYDUINO=129 -DARDUINO_ARCH_AVR -DARDUINO=10609 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy" "-I/Volumes/stuff/numist/Documents/Arduino/libraries/HID/src" "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/Beensy.ino.cpp" -o "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
Using previously compiled file: /var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/ADB.cpp.o
Using previously compiled file: /var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/ADBKeyboard.cpp.o
Using previously compiled file: /var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/ADBMediaKeyboard.cpp.o
Using previously compiled file: /var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/ADBWire.cpp.o
"/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=atmega32u4 -DTEENSYDUINO=129 -DARDUINO_ARCH_AVR -DARDUINO=10609 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy" "-I/Volumes/stuff/numist/Documents/Arduino/libraries/HID/src" "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/Beensy.ino.cpp" -o "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/Beensy.ino.cpp.o"
"/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=atmega32u4 -DTEENSYDUINO=129 -DARDUINO_ARCH_AVR -DARDUINO=10609 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/Volumes/stuff/numist/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy" "-I/Volumes/stuff/numist/Documents/Arduino/libraries/HID/src" "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/USBKeyboard.cpp" -o "/var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/USBKeyboard.cpp.o"
In file included from /Volumes/stuff/numist/Documents/Arduino/libraries/HID/src/HID-Project.h:39:0,
                 from /var/folders/jk/nfqkhlvs11n545w59txlp18w0000gp/T/build449f55bd2d455943cd41a45713cdeba8.tmp/sketch/USBKeyboard.cpp:6:
/Volumes/stuff/numist/Documents/Arduino/libraries/HID/src/SingleReport/SingleAbsoluteMouse.h:28:26: fatal error: PluggableUSB.h: No such file or directory
 #include "PluggableUSB.h"
                          ^
compilation terminated.
Using library HID at version 2.4.3 in folder: /Volumes/stuff/numist/Documents/Arduino/libraries/HID 
Error compiling for board Teensy 2.0.
```

NicoHood, 2016-06-19:
Teensy has its own, different USB core. The error is expected.

Ben-Lichtman, 2017-05-15:
I am also experiencing this PluggableUSB.h error on my teensy 2.0. How can I work around this?

NicoHood, 2017-05-15:
@Ben-Lichtman Its not supported as stated above.

titlyn, 2021-12-11:
I want to make a joystick with my Arduino Wemos D1 & R2 and after including joystick library & compiling , it gives me this type of error:  
In file included from C:\Users\B.O.S.S\Documents\Arduino\libraries\ArduinoJoystickLibrary-version-2.0\src/Joystick.h:24,
                 from C:\Users\B.O.S.S\Desktop\NizGamingTech\NizGamingTech.ino:11:
C:\Users\B.O.S.S\Documents\Arduino\libraries\ArduinoJoystickLibrary-version-2.0\src/DynamicHID/DynamicHID.h:37:12: fatal error: PluggableUSB.h: No such file or directory
   37 |   #include "PluggableUSB.h"
      |            ^~~~~~~~~~~~~~~~
compilation terminated.

NicoHood, 2021-12-12:
You are commenting on the wrong library. This library is not the one you are using