NicoHood/HID #61: "#error HID Project can only be used with an USB MCU."

bobbysq, 2015-10-27:
Whenever I try to compile anything with the library, I get this error:
    In file included from /home/bobby/Arduino/libraries/HID-dev_2_4/examples/BootKeyboard/BootKeyboard.ino:14:0:
    /home/bobby/Arduino/libraries/HID-dev_2_4/src/HID-Project.h:37:2: error: #error HID Project can only be used with an USB MCU.
     #error HID Project can only be used with an USB MCU.
      ^
    exit status 1
    Error compiling.

The sketches I'm trying to use are the example sketches, so it's not an issue with my code. My HID Project version is 4.4 as of c78fc923f4ccb82fb0035bec43189f027739a8e0. My Arduino version is 1.6.6 2015/10/27 05:26. I have HoodLoader installed, and I'm just trying to verify, not upload.

NicoHood, 2015-10-28:
You need to select an Arduino compatible USB board and install the hourly build of the IDE. On top of that atm a PR (that is not merged yet) is required to use the library at the moment. (this should cause another error then but will be solved in the folloing builds).

So you want to check the arduino installation you used and also this issue:
https://github.com/NicoHood/HID/issues/60#issuecomment-150124842 and below

Also make sure your HoodLoader2 installation is up to date. If you are using HoodLoader1, please use the new version 2.

If you solved the issue, please let me know the solution.

bobbysq, 2015-10-28:
I definitely have the right version of Arduino and HL2. Running as root (as suggested in #60) didn't solve anything. If I can, I'll try testing on a Windows computer I have at home.

Is there anything really big that I'll miss if I downgrade to Arduino 1.6.5 and HID Project 2.2?

NicoHood, 2015-10-29:
you miss all new functions and its a lot stabler now.

Running as root is NOT an option. This is wrong, the guy did not read the posts carefully, as you (might) did as well.

You need to delete the avr core inside ./Arduino15/ somewhere. Then it should work. If not, please post a full debug output and the content (folder structure) of ./Arduino15 and your sketchbook.

bobbysq, 2015-10-29:
I don't have the hardware folder. Here's the output of `ls -R`: http://pastebin.com/eSSMxrga

Sketchbook folder: http://pastebin.com/R46xWesh

NicoHood, 2015-10-30:
Huh? Looks all correct to me.

Can you please verify that you selected a HoodLoader2 board? Have you tried to compile for the Leonardo as well? (official Leonardo, not 32u4 from HL2)

I currently havent updated my IDE for a few days, so I cannot check if a new build broke the build process.

The only difference I see is, that you used the package manager to download HoodLoader2. You can also try to place HoodLoader2 inside /sketchbook/hardware/HoodLoader2/. But The Leonardo should work anyways. If the leonardo fails, this wont help anyways.

bobbysq, 2015-11-03:
The Leonardo fails, but differently.   
```
/home/bobby/Arduino/libraries/HID-dev_2_4/src/MultiReport/ImprovedKeyboard.cpp: In member function 'virtual int Keyboard_::send()':
/home/bobby/Arduino/libraries/HID-dev_2_4/src/MultiReport/ImprovedKeyboard.cpp:75:80: error: void value not ignored as it ought to be
  return HID().SendReport(HID_REPORTID_KEYBOARD, &_keyReport, sizeof(_keyReport));
                                                                                ^
exit status 1
Error compiling.
```
I should note that that's with the same hourly build as before. Arduino 1.6.6 is out now and now the console output is:
```
Sketch uses 5,938 bytes (20%) of program storage space. Maximum is 28,672 bytes.
Global variables use 202 bytes (7%) of dynamic memory, leaving 2,358 bytes for local variables. Maximum is 2,560 bytes.
```
It works with the Leonardo. Unfortunately, it still gives the USB MCU error with the HL2 Uno picked.

NicoHood, 2015-11-04:
Can you please also post a full verbose output of the failing HL2 compilation?

With 1.6.6 the avr core also got updated. Maybe those files are in a different place too, idk. Or your build was just too old.

bobbysq, 2015-11-06:
```
/home/bobby/arduino-1.6.6/arduino-builder -dump-prefs -logger=machine -hardware "/home/bobby/arduino-1.6.6/hardware" -hardware "/home/bobby/.arduino15/packages" -tools "/home/bobby/arduino-1.6.6/tools-builder" -tools "/home/bobby/arduino-1.6.6/hardware/tools/avr" -tools "/home/bobby/.arduino15/packages" -built-in-libraries "/home/bobby/arduino-1.6.6/libraries" -libraries "/home/bobby/Arduino/libraries" -fqbn=HoodLoader2:avr:unoHIDBridge -ide-version=10606 -build-path "/tmp/buildb8a321ce41f390adead24b24673a2fe0.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/bobby/Arduino/libraries/HID-dev_2_4/examples/BootKeyboard/BootKeyboard.ino"
/home/bobby/arduino-1.6.6/arduino-builder -compile -logger=machine -hardware "/home/bobby/arduino-1.6.6/hardware" -hardware "/home/bobby/.arduino15/packages" -tools "/home/bobby/arduino-1.6.6/tools-builder" -tools "/home/bobby/arduino-1.6.6/hardware/tools/avr" -tools "/home/bobby/.arduino15/packages" -built-in-libraries "/home/bobby/arduino-1.6.6/libraries" -libraries "/home/bobby/Arduino/libraries" -fqbn=HoodLoader2:avr:unoHIDBridge -ide-version=10606 -build-path "/tmp/buildb8a321ce41f390adead24b24673a2fe0.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/bobby/Arduino/libraries/HID-dev_2_4/examples/BootKeyboard/BootKeyboard.ino"
"/home/bobby/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bobby/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/bobby/arduino-1.6.6/hardware/arduino/avr/variants/standard" "/tmp/buildb8a321ce41f390adead24b24673a2fe0.tmp/sketch/BootKeyboard.ino.cpp"
"/home/bobby/arduino-1.6.6/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bobby/arduino-1.6.6/hardware/arduino/avr/cores/arduino" "-I/home/bobby/arduino-1.6.6/hardware/arduino/avr/variants/standard" "-I/home/bobby/Arduino/libraries/HID-dev_2_4/src" "/tmp/buildb8a321ce41f390adead24b24673a2fe0.tmp/sketch/BootKeyboard.ino.cpp"
In file included from /home/bobby/Arduino/libraries/HID-dev_2_4/examples/BootKeyboard/BootKeyboard.ino:14:0:
/home/bobby/Arduino/libraries/HID-dev_2_4/src/HID-Project.h:37:2: error: #error HID Project can only be used with an USB MCU.
 #error HID Project can only be used with an USB MCU.
  ^
Using library HID-dev_2_4 at version 2.4 in folder: /home/bobby/Arduino/libraries/HID-dev_2_4 
exit status 1
Error compiling.
```

And just to confirm, I am uploading to *HoodLoader 2 Uno*. The error doesn't happen when I compile for a 16u2, or anything else with native USB capabilities.

NicoHood, 2015-11-06:
Guess why? Because the Uno has no USB capabilities and the 16u2 does. Thatswhy HoodLoader2 was invented, to use the 16u2 instead of the 328 (aka Uno in this context).

I guess this can be closed than?

bobbysq, 2015-11-06:
I guess so. Sorry for wasting your time with this. I'll just upload to the 16u2 and if I need to use all the Uno pins, I'll use HID Bridge.

closed 2015-11-06 by bobbysq

NicoHood, 2015-11-06:
Yep. You just need to develop your own Bridge since I do not have the time to and abandoned this work. But you are still able to include the HID APIs which do a lot of work for you. You then just need a carrier for your data and a flush function (to not send the data too fast). Or you maybe invent something more simpler.