NicoHood/HID #62: HID too big for 8u2?

flupes, 2015-10-30:
Thanks to your project, I was really excited to try giving a new life to an old Uno-328 with a mega8u2.
I looks however that I cannot fit the code on the tiny 8u2 (I am using dev_2_4 rolled back to #357baf2)

Is there an option for a "lite" version (looks like 1.7 had something)? I am doing something really wrong (I am worried about the 2 versions of the HID listed)?

Partial build log:

Arduino: 1.6.6 Hourly Build 2015/10/29 11:41 (Windows 7), Board: "HoodLoader2 8u2, HoodLoader2 Uno"

R:\Arduino\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware "R:\Arduino\arduino-nightly\hardware" -hardware "C:\Users\flupes\AppData\Local\Arduino15\packages" -tools "R:\Arduino\arduino-nightly\tools-builder" -tools "R:\Arduino\arduino-nightly\hardware\tools\avr" -tools "C:\Users\flupes\AppData\Local\Arduino15\packages" -libraries "R:\Arduino\arduino-nightly\libraries" -libraries "C:\Users\flupes\devel\arduino\libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega8u2:board=uno -ide-version=10606 -build-path "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\flupes\devel\arduino\libraries\HID-project\examples\System\System.ino"
R:\Arduino\arduino-nightly\arduino-builder -compile -logger=machine -hardware "R:\Arduino\arduino-nightly\hardware" -hardware "C:\Users\flupes\AppData\Local\Arduino15\packages" -tools "R:\Arduino\arduino-nightly\tools-builder" -tools "R:\Arduino\arduino-nightly\hardware\tools\avr" -tools "C:\Users\flupes\AppData\Local\Arduino15\packages" -libraries "R:\Arduino\arduino-nightly\libraries" -libraries "C:\Users\flupes\devel\arduino\libraries" -fqbn=HoodLoader2:avr:HoodLoader2atmega8u2:board=uno -ide-version=10606 -build-path "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\flupes\devel\arduino\libraries\HID-project\examples\System\System.ino"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega8u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484E '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 8u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16  "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\sketch\System.ino.cpp"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega8u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484E '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 8u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\flupes\AppData\Local\Arduino15\packages\HoodLoader2\hardware\avr\2.0.5\variants\HoodLoader2" "-IC:\Users\flupes\devel\arduino\libraries\HID-project\src" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\sketch\System.ino.cpp"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega8u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484E '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 8u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\flupes\AppData\Local\Arduino15\packages\HoodLoader2\hardware\avr\2.0.5\variants\HoodLoader2" "-IC:\Users\flupes\devel\arduino\libraries\HID-project\src" "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\HID" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\sketch\System.ino.cpp"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega8u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484E '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 8u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\flupes\AppData\Local\Arduino15\packages\HoodLoader2\hardware\avr\2.0.5\variants\HoodLoader2" "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\HID" "-IC:\Users\flupes\devel\arduino\libraries\HID-project\src" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\sketch\System.ino.cpp"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega8u2 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_HOODLOADER2 -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x484E '-DUSB_MANUFACTURER="Nico Hood"' '-DUSB_PRODUCT="HoodLoader2 8u2"' -DMAGIC_KEY_POS=(RAMEND-1) -DUSB_EP_SIZE=16 "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\flupes\AppData\Local\Arduino15\packages\HoodLoader2\hardware\avr\2.0.5\variants\HoodLoader2" "-IC:\Users\flupes\devel\arduino\libraries\HID-project\src" "-IR:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\HID" "C:\Users\flupes\devel\arduino\libraries\HID-project\src\MultiReport\AbsoluteMouse.cpp"

...

"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=atmega8u2  -o "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/System.ino.elf" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\sketch\System.ino.cpp.o" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\libraries\HID-project\HID-project.a" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp\libraries\HID\HID.cpp.o" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/core\core.a" "-LC:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp" -lm
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/System.ino.elf" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/System.ino.eep"
"R:\Arduino\arduino-nightly\hardware\tools\avr/bin/avr-objcopy" -O ihex -R .eeprom  "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/System.ino.elf" "C:\Users\flupes\AppData\Local\Temp\build8e98f594d9c88f68b9b1895d65446e20.tmp/System.ino.hex"
Using library HID-project at version 2.4 in folder: C:\Users\flupes\devel\arduino\libraries\HID-project 
Using library HID at version 1.0 in folder: R:\Arduino\arduino-nightly\hardware\arduino\avr\libraries\HID 

Sketch uses 5,394 bytes (131%) of program storage space. Maximum is 4,096 bytes.
Global variables use 157 bytes (30%) of dynamic memory, leaving 355 bytes for local variables. Maximum is 512 bytes.
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
	at cc.arduino.Compiler.size(Compiler.java:312)
	at cc.arduino.Compiler.build(Compiler.java:153)
	at processing.app.Sketch.build(Sketch.java:1108)
	at processing.app.Sketch.build(Sketch.java:1083)
	at processing.app.Editor$BuildHandler.run(Editor.java:2011)
	at java.lang.Thread.run(Thread.java:745)
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

NicoHood, 2015-10-30:
Well in this case there is no way. Normally the UNO R3 have a 16u2 which is quite good.

The 8u2 is somehow more limited in flash. The bootloader takes 4/8kb. And the usb Core itself is around 4kb.

So for the nearer future there are only a few things that could work:
* Dont use USB at all (deactivation support is pending via a PR in the Arduino Project).
* Deactivate CDC Serial (needs to be patched in the Arduino core, will be a bit more complex, isnt done yet, maybe you can do this?).
* Use a LUFA project (without the IDE) which are in general more stable/advanced, but you dont have the Arduino IDE and its libraries.
* Dont use a bootloader and burn via ISP directly (might not be what you want for debugging etc.)

I'd go for Serial deactivation. I dont have the time to, but some people really would like to see this.

flupes, 2015-10-30:
Thanks for providing these possible options!
I may look into the CDC Serial, but I am not sure it is worth the effort since it seems I have a very unique case due to this very old board (it is an official Uno first revision).

closed 2015-10-30 by flupes