NicoHood/HID #66: Missing HID macro definitions

obra, 2015-12-01:
If I revert a95361043, I'm now seeing missing definitions: 

```
/Users/jesse/Documents/Arduino/libraries/HID/src/SingleReport/BootKeyboard.cpp: In member function 'virtual bool BootKeyboard_::setup(USBSetup&)':
/Users/jesse/Documents/Arduino/libraries/HID/src/SingleReport/BootKeyboard.cpp:150:24: error: 'HID_REPORT_TYPE_FEATURE' was not declared in this scope
    if(setup.wValueH == HID_REPORT_TYPE_FEATURE){
                        ^
/Users/jesse/Documents/Arduino/libraries/HID/src/SingleReport/BootKeyboard.cpp:165:29: error: 'HID_REPORT_TYPE_OUTPUT' was not declared in this scope
    else if(setup.wValueH == HID_REPORT_TYPE_OUTPUT){
                             ^
/Users/jesse/Documents/Arduino/libraries/HID/src/SingleReport/BootKeyboard.cpp:173:29: error: 'HID_REPORT_TYPE_INPUT' was not declared in this scope
    else if(setup.wValueH == HID_REPORT_TYPE_INPUT)
                             ^
exit status 1

```

It looks almost like there's a missing header file that just never got checked in. Grepping for HID_REPORT_TYPE_ , I see checks but no definitions.

NicoHood, 2015-12-02:
OH yeah, they did not merge my PR yet:
https://github.com/arduino/Arduino/pull/4105

Test it and comment that they should merge it.

obra, 2015-12-02:
>    OH yeah, they did not merge my PR yet:
>    [1]arduino/Arduino#4105
>
>    Test it and comment that they should merge it.

I'm confused about why the code change to add long recv support and the
new HID constants are in the same pull req. 

(I don't have anything that would want/need the long recv support)

NicoHood, 2015-12-03:
Both was needed for the RAWHID. And since they almost instantly merged all my PRs before, I thought they would do it now again, since the change is so minor. Comment there if its fine for you to merge.

NicoHood, 2015-12-12:
Is it now working with the new hourly build? I still need to fix the >64 byte rawhid reads but it should compile fine still

NicoHood, 2015-12-19:
Fixes with IDE 1.6.7 and 2.4.1 HID Project.
Could zou please check if it all works fine?

closed 2015-12-19 by NicoHood