NicoHood/HID #68: Modifier keys not working properly on Leonardo

Lexazan, 2016-01-04:
Hi, I can't get several modifies to work, e.g. Alt + Shift + Tab. After sending second modifier the first one releases. The standart arduino keyboard library is working as expected.

Code sample:
```
  Keyboard.press(KEY_LEFT_ALT);
  Keyboard.press(KEY_LEFT_SHIFT);
  Keyboard.press(KEY_TAB);
  Keyboard.releaseAll()
```

IDE 1.6.7, HID Project 2.4.1
Board: Sparkfun Pro Micro MEGA32U4 (Arduino Leonardo comparable)

NicoHood, 2016-01-04:
Very good catch!

Does this now fix it?
https://github.com/NicoHood/HID/commit/f96e09c75b32e67df5fe4a964b8e7ad29af7eebe

You can use the dev branch with IDE 1.6.8 to test this. However this patch should also work with the 1.6.7 IDE (but you need to remove the compiler error manually).

Lexazan, 2016-01-04:
Can confirm multiple modifiers working after this patch! Thank you for great library and quick fix :)

closed 2016-01-04 by Lexazan

NicoHood, 2016-01-04:
Release it out there.
https://github.com/NicoHood/HID/releases/tag/2.4.2

Thanks for reporting though ;)