NicoHood/HID #71: Add V-USB support

NicoHood, 2016-01-14:
It would be nice to also support non native usb avrs with V-USB.
Popular devices are the watterott nanite series or the adafruit trinket products.

leobel96, 2018-03-09:
@NicoHood are there news about it?

leobel96, 2018-03-09:
@NicoHood maybe [this](https://github.com/adafruit/Adafruit-Trinket-USB/tree/master/TrinketHidCombo) library that also support Digispark may help or [this](https://github.com/digistump/DigisparkArduinoIntegration/tree/master/libraries/DigisparkKeyboard) one

NicoHood, 2018-03-10:
I am not interested in using vusb myself, so I will not implement it. Also I am trying to build a more stable usb keyboard using lufa instead currently. https://github.com/NicoHood/avr/tree/master/lib/USB_KEYBOARD

I dont know how stable vusb is, but there are reasons why AVRs with dedicated hardware exists. Also its way better to also have a usb CDC serial port. So nope, no progress on my side.