NicoHood/HID #73: First byte missing on Win10

DeqingSun, 2016-01-18:
[image]

I suspect this is related to descriptor or driver.
The same code works well on Mac 10.10.5

And the same Java API works well on Win10 with my other PIC16F1459 project.

NicoHood, 2016-01-19:
Wait.
This is **your** code, not mine. I know where the issue exists. If it was you on tox lately, you can also hit me there to chat.

Explaining it a bit:
Normally HID devices work with multireports:
https://github.com/NicoHood/HID/wiki/API-Documentation#single-and-multi-report-hid-devices

RawHID conflicts with other devices inside the same report. This does not mean, that you can have two single report devices though. However **if** you have a  multireport device, the first byte is used to determine which device in the report is used (device id). Normally this is only important for sending data from Arduino to the PC.

I have not that much experience with the opposite (sending to the Arduino), but I guess you API messes this up. Since processing is just a function wrapper of some APIs, this can be different from Windows to mac. You might need to check the related API.

If I remember right, the reportID for an out report is stored in some of the metadata bytes (setup struct):
https://github.com/NicoHood/HID/blob/master/src/SingleReport/RawHID.cpp#L84

Also as you can see in your screen shot, this is a windows problem, not an arduino problem. However I find your work very interesting. I had no time to check your PR yet (remind me if I forget about it). I really would like to add processing examples, and if you like to, you can write additional texts for the wiki.

DeqingSun, 2016-01-19:
You are right. My PIC device has a reportID in descriptor. I did another test with same code on a Win7 32bit machine. I got these result:
[image]
[image]
If the first byte is 0, it will be chopped within Windows. Otherwise, it triggers a IOException.

I also found a stackoverflow question related to this problem:
https://stackoverflow.com/questions/22240591/confusing-of-report-id-when-using-hidapi-through-usb/22660386#22660386

NicoHood, 2016-01-19:
Yeah, as i said. And 0 indicates no report ids are used, since 1 is the first one. You should add something like an #idef or so. Also it looks like that those functions are just libusb wrappers. So they might be coded badly/not the same way. So the API failed (might not be you fault though).

Feel free to report any further progress.

DeqingSun, 2016-01-19:
I got it working by a workaround in PR.
I think this processing is helpful for using HIDRaw since I never got HIDAPI compiled on windows for years. Other people may have the same problem.

closed 2016-06-26 by NicoHood