NicoHood/HID #78: Key modifiers not working on Leonardo

paulnd, 2016-03-04:
I'm using v2.4.3 and alpha and numerical keys work correctly but all key modifiers display as an ascii key, for example KEY_BACKSPACE displays as * KEY_F1 displays as : etc

NicoHood, 2016-03-05:
can you please show the code?

paulnd, 2016-03-05:
Here is the code, apologies if it is my error, I'm still learning
```
#include <HID-Project.h>

const int ExecLED = LED_BUILTIN;
const int COLS = 12;
const int ROWS = 6;

int MatrixColumnPins[12] = {12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1}; //connect to the column pinouts of the keypad
int MatrixRowPins[6] = {23, 22, 21, 20, 19, 18}; //connect to the row pinouts of the keypad

char MatrixScancodes[COLS][ROWS] = {
  {KEY_F1,        KEY_F2,   KEY_F3,         KEY_F4,       KEY_F5,     KEY_F6},
  {KEY_F12,       KEY_F11,  KEY_F10,        KEY_F9,       KEY_F8,     KEY_F7},
  {']',           KEY_QUOTE,  '[',          ',',          KEY_INSERT,   0},
  {KEY_END,       KEY_ENTER,  KEY_BACKSLASH, '*',          '+',        KEY_HOME},
  {'J',           'E',        'O',          'T',          'Y',        KEY_BACKSPACE},
  {'I',           'D',        'N',          'S',          'X',        '/'},
  {'H',           'C',        'M',          'R',          'W',        KEY_INSERT},
  {'G',           'B',        'L',          'Q',          'V',        KEY_SPACE},
  {'F',           'A',        'K',          'P',          'U',        'Z'},
  {KEY_PAGE_DOWN, ';',        '3',          '6',          '9',        '-'},
  {KEY_PAGE_UP,   KEY_TILDE,  '2',          '5',          '8',        '0'},
  {'n',           'p',        '1',          '4',          '7',        '.'}
};

int MatrixPreviousState[12][6];      // previous state of scanning
int dstate;

void setup()
{
  pinMode(ExecLED, OUTPUT);

  delay(4000); // time to reprogram the board if something is going wrong

  BootKeyboard.begin();
}

void loop()

{
  //LED test

  if (BootKeyboard.getLeds() & LED_CAPS_LOCK)
    digitalWrite(ExecLED, HIGH);
  else
    digitalWrite(ExecLED, LOW);

  { // init keyboard rows
    for (byte r = 0; r < ROWS; r++) {
      pinMode(MatrixRowPins[r], INPUT_PULLUP);
    }
    // scan the matrix
    for (int c = 0; c < COLS; c++) {
      pinMode(MatrixColumnPins[c], OUTPUT);
      digitalWrite(MatrixColumnPins[c], LOW);
      for (int r = 0; r < ROWS; r++) {
        dstate = digitalRead(MatrixRowPins[r]);
        if (dstate == LOW && MatrixPreviousState[c][r] == 1) // if key is newly pressed
        {
          BootKeyboard.write(MatrixScancodes[c][r]);
        }
        if (dstate == HIGH && MatrixPreviousState[c][r] == 0) // if key is just released
        {
          BootKeyboard.release(MatrixScancodes[c][r]);
        }
        MatrixPreviousState[c][r] = dstate;
      }    // for rows
      // high impedance input
      digitalWrite(MatrixColumnPins[c], HIGH);
      pinMode(MatrixColumnPins[c], INPUT);
    } // for cols
  }
}
```

NicoHood, 2016-03-05:
You need to explicitly make it a modifier key then:
https://github.com/NicoHood/HID/blob/master/examples/Keyboard/ImprovedKeyboard/ImprovedKeyboard.ino#L42-L43

I know the docs lack some information here, but I do not have time for that.
So you have no chance to get it working like that.

For understanding:
You have 2 different keymaps. ASCII and real USB Keymaps.
Ascii is uint8_t and usb is KeyboardKeycode. See those two implementations:
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/KeyboardAPI.hpp#L149
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/DefaultKeyboardAPI.hpp#L28

And in your matrix you save a mix of both, and they will result in uint8_t values (char is wrong).
You rather have to use one of each, but not both at the same time.

So in order to get your idea somehow working you need to use the direct usb scancodes. Those definitions can be found here:
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/ImprovedKeylayouts.h#L32

But with this no shift + a (aka 'A') is possible. That is the limitation. You could code your own functions/mapping tables around this though.

closed 2016-03-05 by NicoHood