NicoHood/HID #81: Android MUTE

adrianrb, 2016-04-06:
Hi!

I´m trying to use a IR remote to control an Android stick (simple commands, UP, DOWN, VOL+, VOL- etc).

But I can´t do MEDIA MUTE (**Consumer.write(MEDIA_VOLUME_MUTE)** doesn´t works).

How can I do it? (using this library).

Adrian

Some info: 
Android keys: https://source.android.com/devices/input/keyboard-devices.html

NicoHood, 2016-04-07:
Your android device down not support the consumer HID device. Does the other consumer keys work?
You can try to replace consumer with `Keyboard.write(KEY_VOLUME_MUTE);`

Can you try the sketch on a pc and see if the pc does mute the sound?

adrianrb, 2016-04-07:
Hi Nico.

Tks for the answer.

Vol+ and Vol- works, in PC **and** Android (and Power too - in android), but your suggestion `Keyboard.write(KEY_VOLUME_MUTE);` does not works (both PC and android).

I will keep trying (any advice will be appreciated).

Adrian

NicoHood, 2016-04-07:
Does volume mute work on pc (using the consumer device)?

adrianrb, 2016-04-07:
Yes, in PC works, but in android nop (tested again):

`Consumer.write(MEDIA_VOLUME_MUTE);`

NicoHood, 2016-04-07:
Then it is a problem of your android device. Try to find another keyboard command, maybe there is something else. Otherwise I am sorry that there is no other solution.

closed 2016-04-07 by NicoHood