NicoHood/HID #82: Is this library meant to be extended?

ekr1, 2016-06-15:
Hello,

I would like to read the "11 LED Page - 11.3 Consumer Indicators - Mute" indicator; or I would like to use "15.9.1 Volume (LC)" which is not a simple keypress but has data associated with it.

This would mean to extend an existing usage table (or create a new endpoint/device). Is the NicoHood-HID library intended to be my "base point" for this, or is it not the right thing for such an extension? Am I better off creating my own library for this, using the Arduino core HID classes directly?

I guess this is simply a question and not really an "issue"... maybe a "documentation issue". ;) Feel free to delete/close this if it is not the right place here.

Thank you!

NicoHood, 2016-06-15:
Feel free to adapt it to this project and open a PR.
But I think i already tested all leds and not all are supported by the OS. Are you sure this led will even work?

ekr1, 2016-06-15:
I have no idea if it would work, no, but I'd be interested in trying (it was more of an example). The other one (the volume with the absolute value) would be even more interesting for me.

I'm currently more or less fooling around with the Arduino USB stuff, and digging into the USB/HID specifications as well for learning purposes (my end goal is to build some flightsim control boxes with lots and lots of controls of all kinds - getting my hands wet with the volume stuff first ;) ).

closed 2016-06-26 by NicoHood