NicoHood/HID #83: BootKeyboard sends wrong keys

obfuscated, 2016-06-18:
I'm trying to use a SingleReport/BootKeyboard but it sends wrong key codes. I'm using a Micro and Arduino 1.6.9, gcc 4.9.x.

The problem is that if I send KEY_ENTER the computer receives '('.
I have BootKeyboard.press(KEY_ENTER) in the code.

I've tried to trace the problem and it turns out that the KeyboardAPI::set expects an index in the _asciimap, but it gets a KeyboardKeycode value instead. So it expects 10, but the code for KEY_ENTER is 40 and this is exactly the ascii for '('.

If I comment the line: `k = pgm_read_byte(_asciimap + k);` it starts to work, but now all the modifiers doesn't work (shift, control, alt and the windows button).

Am I doing something wrong or the code is just broken?

obfuscated, 2016-06-18:
For the modifiers I'm using KeyboardKeycode values of: KEY_RIGHT_ALT, KEY_LEFT_ALT, KEY_RIGHT_CTRL, KEY_LEFT_CTRL, KEY_RIGHT_SHIFT, KEY_LEFT_SHIFT, KEY_LEFT_GUI.

NicoHood, 2016-06-18:
I do not know why this happens, but there is no pgm_read_byte() in the set() function:
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/DefaultKeyboardAPI.hpp#L53

So where is the code you are talking about?

obfuscated, 2016-06-19:
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/KeyboardAPI.hpp#L157

NicoHood, 2016-06-19:
The (expected) chain works like this:

You call [press()](https://github.com/NicoHood/HID/blob/master/src/HID-APIs/KeyboardAPI.hpp#L55) which calls [add()](https://github.com/NicoHood/HID/blob/master/src/HID-APIs/KeyboardAPI.hpp#L77) which calls set().

Now the there are **two** set() functions. One with uint8_t as param and one with the enum. But the one with the enum is virtual only and needs to be overriden:
https://github.com/NicoHood/HID/blob/master/src/HID-APIs/KeyboardAPI.h#L59

So there are only two possible reasons for this bug:
* The press() chain calls the wrong set() function
* The wrong press() function is called which of course calls the "wrong" set()

To debug this:
* Can you try [this](https://github.com/NicoHood/HID/blob/master/examples/Keyboard/BootKeyboard/BootKeyboard.ino#L38) example and check if it works?
* Can you replace the marked line with a press() and a release() instead? This should trigger your bug
* If this works, can you give me a **minimal** test sketch where you can reproduce the bug?

I have not tested anything myself, because all my hardware is in use or broken ;)

obfuscated, 2016-06-19:
OK, Seems to be erroneous usage. I was using an uint8_t type and I didn't see the other overload of the set function.

closed 2016-06-19 by obfuscated

NicoHood, 2016-06-19:
You're welcome. I know this is confusing a lot, but that is the simplest way to support all methods and also the current Arduino implementation.

obfuscated, 2016-06-19:
I'm not sure this flexibility is needed, but anyway this is your decision.

This problem is another example where using function overloading is a bad decision, especially when the types are so close - uint8_t and enum A:uint8_t.

Also all these virtual calls seems like a really bad decision. I hope you know that most compilers cannot de-virtualize them. The only compiler that is advertised as capable to do it is gcc>=5.x and only in LTO mode. All the others fail pretty miserably. And also they don't respect final too much. :)
Your library seems like a good case where the CRTP pattern can be applied.

NicoHood, 2016-06-19:
I used CRTP a while ago and you might be right. At the time of writing the code I did not know that, so I might try that out when I know more and got the time. Code is always improving, thanks for the suggestion.

With de-virtualizing you mean that an overhead is generated which is not really required? I am aware of this, but otherwise the library would be very complex to maintain. It was a decision i made for now.