NicoHood/HID #88: Gamepad example not sending input on Windows 10

swoobie, 2016-08-22:
Hey NicoHood,
I'm having trouble getting Windows 10 to recognize any input using the Gamepad example.
I'm running an Arduino Uno with a 16u2 and Hoodloader 2. I've uploaded the Gamepad example sketch and the Uno shows up in the devices menu as seen in the following:
[image]

I attached a button to the PB2 pin for the 16u2 ( I believe that's the correct pin '2' used in the example ) and the onboard LED does switch off when the button is pressed. Any advice?

dragonet80, 2016-08-23:
Where is the button tied to? Ground or Vcc?

NicoHood, 2016-08-23:
Yes this is the normal upload behavior. I've seen you added me in tox, just ask there other questions ;)

closed 2016-08-23 by NicoHood