NicoHood/HID #90: Rename dPad(1|2) to hat(1|2)?

RandomInsano, 2016-10-23:
So, I built a USB arcade joystick with this library, and it went *[fantastically](https://github.com/RandomInsano/ArcadeMarkI)* well. There is one problem though when I go to play Street Fighter: the dpad function doesn't give me anything.

It turns out there's a subtle difference in the USB HID spec that differentiates a flight "hat" from a "direction pad". It's outlined pretty clearly in ([section 4.7 of HID Usage Tables](http://www.usb.org/developers/hidpage/Hut1_12v2.pdf). So, that's fun.

All this to say, it might be good to alias the dPad# functions to hat# and make a plain dPad() function? I need to see how to implement the actual four directions which seem to be their own device class each:

```
	0x90	D-pad Up
	0x91	D-pad Down
	0x92	D-pad Right
	0x93	D-pad Left
```
Taken from http://www.freebsddiary.org/APC/usb_hid_usages.php

I don't think it'll be hard, but this is kind of magic to me at the moment. Would someone be willing to provide some help and pointers? Generating a new device report is going to be the big one.

RandomInsano, 2016-10-23:
Hmm... Noticed that the dpad events are system-wide and provided in the "System API"? This seems a bit strange, but I'll experiment to see if it makes a difference to this game or not.

RandomInsano, 2016-10-23:
Update from above, system-wide dpad usage *seems* like it was added accidentally. There is reference of someone using it successfully [here](http://www.avrfreaks.net/forum/d-pad-vs-buttons), and I think it was in the gamepad report, not the system one.

From some [reverse engineering of the Xbox360 controller](http://euc.jp/periphs/xbox-pad-report-desc.txt), it's using a Pointer type with four axes and a single bit of precision, but it does seem to use those hex values I referenced above.

```
			; D-pad
0501			Usage Page (Generic Desktop)
0901			Usage (Pointer)
A100			Collection (Physical)
7501				Report Size (1)
1500				Logical Minimum (0)
2501				Logical Maximum (1)
3500				Physical Minimum (0)
4501				Physical Maximum (1)
9504				Report Count (4)
0501				Usage Page (Generic Desktop)
0990				Usage (D-pad Up)
0991				Usage (D-pad Down)
0993				Usage (D-pad Left)
0992				Usage (D-pad Right)
8102				Input (Data,Variable,Absolute)
C0			End Collection
```

RandomInsano, 2016-10-23:
After a huge amount of research and playing around with this, I have discovered two things:

1. The XBox360's pointer definition doesn't read properly for other devices and shows up as four additional buttons via an SDL test app I have (32 to 35).
2. Street Fighter just wants the *other* hat! 😆  

Removing the existing hat definitions does not allow the Pointer type to show up as a directional button, but the fact that the usage reports *may* actually be mapping correctly seems promising... I'm going to fork and keep this in a branch, but I don't think it makes sense to port to this project since I can't find a use for all this research.

Learned a lot about HID at least. Sorry for the spam.

closed 2016-10-23 by RandomInsano

NicoHood, 2016-10-23:
It would be intertesting to show what you've changed, so others can see your patches.

The XBOX controller is not an HID controller and something mcirosoft proprietary. Some games only support XBOX Controller, some do support the normal hid controllers. Depending on the game you could also just use buttons for the direction sometimes.

RandomInsano, 2016-10-24:
Here's the commit:
https://github.com/RandomInsano/HID/commit/319bb4ee6f646b96f239008d306e94451547069a

Only really relevant change is the device now reports as Gamepad.

NicoHood, 2016-10-24:
Oh I remember that all other solutions online used a joystick. I am not sure why i switched back to joystick (as i first used gamepad) but there was a reason that something was not working with the gamepad class.

RandomInsano, 2016-10-25:
Huh... mighty interesting. In any event, dPad buttons certainly didn't come up right at all, so not overly helpful. Might need to play with that more.