NicoHood/HID #91: RawHID.write not working on a Leonardo clone

jh1-git, 2016-11-02:
Using the /examples/RawHID/RawHID/RawHID.ino sketch on a Leonardo clone board with a LED and Button attached. Using the test program in /extras/rawhid/ to communicate with it.
I am able to send data from the PC to the board - the USB LED and LED flash.
When I press the button the LED flashes but the USB LED doesn't flash and nothing is received by the PC.  I have also used Wireshark to sniff the USB traffic and it show there is no communication when I press the Button.

NicoHood, 2016-11-05:
Arduino IDE 1.6.9 was reported to work and creashed with 1.6.10.

jh1-git, 2016-11-06:
Looks like this change makes a RawHID.write of 64 bytes stop working.
https://github.com/arduino/Arduino/pull/4864

jh1-git, 2016-11-06:
As a workaround you can either revert the arduino IDE change referenced above, which may leave you with the issue they were trying to fix in CDC and other modes, 
or you can change \libraries\HID\src\SingleReport\RawHID.h in line 50 from
#define RAWHID_TX_SIZE RAWHID_SIZE
to
#define RAWHID_TX_SIZE 63
or some other value under 64, and then, when you call RawHID.write, send this smaller block size.

NicoHood, 2016-12-16:
It also broke the gamepad. Vote the bug here: https://github.com/arduino/Arduino/issues/5732

mnhpias, 2017-03-10:
I'm unable to work RawHID.write , 
I changed  \libraries\HID\src\SingleReport\RawHID.h in line 50 from #define RAWHID_TX_SIZE RAWHID_SIZE to #define RAWHID_TX_SIZE 63  , then I sent 50 bytes long uint8_t throgh RawHID.write , but pc didn't receive anything . pls help me to make  RawHID.write working .

jbutzine, 2017-04-06:
Has a solution been identified for this problem?  Please update this thread if that is the case.

RawHID.write does not work.  No data is sent to the PC.

NicoHood, 2017-04-06:
@jbutzine Which IDE version do you use? It should be fixed in 1.8.2 but I never tested it. And I have no time in the near future

jbutzine, 2017-04-06:
I'm using 1.8.2

jbutzine, 2017-04-06:
Ok.  It works.  I have to back out a few changes I tweaked to determine what change made it work.  I'll update this thread.

jbutzine, 2017-04-06:
The call to RawHID.write needs to be :   RawHID.write(buffer, (sizeof(buffer)-1));
Perhaps this is only when the buffer size is 64.
Now that the code will actually send data, any thoughts on how I can still send 64 bytes instead of 63?

NicoHood, 2017-04-06:
Then the bug is still inside the code or arduino core. Someone needs to look at it and fix it. I dont have time nor equipment now.

jbutzine, 2017-04-20:
My apologies; it does appear that this bug was fixed with the 1.8.2 release of Arduino.  My build was still using the old USBCore.cpp.  When doing the Arduino update the install updated the USBCore.cpp located here:  C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\USBCore.cpp

Instead of the one here: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.17\cores\arduino\USBCore.cpp

When I manually updated the file in the AppData directory, then the 64 byte transactions worked.  Sorry for the boondoggle.

closed 2017-04-23 by NicoHood

IgorIemeluyanofflyanov, 2021-04-04:
Thank to authors ! I change c:\Users\\Documents\Arduino\libraries\HID-Project\src\SingleReport\RawHID.h to 
#define RAWHID_TX_SIZE 64 and all work fine! Arduino 1.8.13 + HoodLoader2.0.5+ Arduino UNO  (at16u2)