NicoHood/HID #92: typeKey(KEY_RETURN) types "(" instead of {ENTER}

CiriousJoker, 2016-11-02:
The headline sums it up pretty well, but I'll further elaborate:

I tried this using a German and the US keyboard layout, but after I looked at the files "TeensyKeylayouts.h" and "ImprovedKeylayouts.h", I noticed that KEY_ENTER always returns 40, which is, - i suppose - an ASCII value.

And that Ascii value translates to "(".

I don't quite get if I'm missing out anything or if that's really a bug no one ever noticed before me ...

NiKiZe, 2016-11-03:
You can see a similar discussion and my response at: https://github.com/NicoHood/HID/issues/89#issuecomment-246488906

Keyboards sends keyboard scancodes, there is no concept of ASCII or any type of char, it is just a code for the pressed key.

On page 54 of http://www.usb.org/developers/hidpage/Hut1_12v2.pdf you see scancode 40 in decimal is Enter

NicoHood, 2016-11-03:
German layout is not supported and covered in another issue. Feel free to implement it.

closed 2016-11-03 by NicoHood

donid, 2016-11-03:
For German layout support please see [my comment in issue 22](https://github.com/NicoHood/HID/issues/22#issuecomment-248208465)

CiriousJoker, 2016-11-03:
I solved the problem by not using your library and changing the Keyboard.cpp instead.

These are the changes I made and from what I've noticed, it all works pretty well including the ALTGR key on the german keyboard (it's used for these symbols: €²³{[]}\~|)

<details> 
  <summary>**My custom _asciimap[]**</summary>
```
extern

// NOTE: Was [128] before
const uint8_t _asciimap[256] PROGMEM;

#define SHIFT 0x80
#define ALTGR 0x40
const uint8_t _asciimap[256] =
{
	0x00,             // NUL
	0x00,             // SOH
	0x00,             // STX
	0x00,             // ETX
	0x00,             // EOT
	0x00,             // ENQ
	0x00,             // ACK  
	0x00,             // BEL
	0x2a,			// BS	Backspace
	0x2b,			// TAB	Tab
	0x28,			// LF	Enter
	0x00,             // VT 
	0x00,             // FF 
	0x00,             // CR 
	0x00,             // SO 
	0x00,             // SI 
	0x00,             // DEL
	0x00,             // DC1
	0x00,             // DC2
	0x00,             // DC3
	0x00,             // DC4
	0x00,             // NAK
	0x00,             // SYN
	0x00,             // ETB
	0x00,             // CAN
	0x00,             // EM 
	0x00,             // SUB
	0x00,             // ESC
	0x00,             // FS 
	0x00,             // GS 
	0x00,             // RS 
	0x00,             // US 

	0x2c,		   //  ' '
	0x1e|SHIFT,	   // !
	0x1f|SHIFT,	   // "
	0x31,          // #
	0x21|SHIFT,    // $
	0x22|SHIFT,    // %
	0x23|SHIFT,    // &
	0x31|SHIFT,    // '
	0x25|SHIFT,    // (
	0x26|SHIFT,    // )
	0x30|SHIFT,    // *
	0x30,          // +
	0x36,          // ,
	0x38,          // -
	0x37,          // .
	0x24|SHIFT,    // /
	0x27,          // 0
	0x1e,          // 1
	0x1f,          // 2
	0x20,          // 3
	0x21,          // 4
[... 295 lines cut ...]
		// ENDNOTE

	 }
	
	// Test the key report to see if k is present.  Clear it if it exists.
	// Check all positions in case the key is present more than once (which it shouldn't be)
	for (i=0; i<6; i++) {
		if (0 != k && _keyReport.keys[i] == k) {
			_keyReport.keys[i] = 0x00;
		}
	}

	sendReport(&_keyReport);
	return 1;
}
```
</details>

I have no idea how this would work together with your library, but I also don't have the time to completely read into it.

I just thought this might be helpful for you, so here you go, maybe you'll find a way to correctly implement it.

Here's the complete Keyboard.cpp file, currently you'd have to put it in the .ino directory and #include it directly.

[Keyboard.cpp](https://github.com/NicoHood/HID/files/570306/Keyboard.zip)

scpcom, 2017-04-19:
For AltGr and special case 0x64 you do not need to modify the Keyboard the library.
Example:
```
#include "Keyboard.h"

#define KEY_ALT_GR KEY_RIGHT_ALT

void setup()
{
  Keyboard.begin();
}

void loop()
{
  if (doIt)
  {
    // special case 0x64 (0xEC = 0x88 + 0x64)
  
    // <
    Keyboard.press(0xEC);
    Keyboard.releaseAll();

    // >
    Keyboard.press(KEY_LEFT_SHIFT);
    Keyboard.press(0xEC);
    Keyboard.releaseAll();

    // |
    Keyboard.press(KEY_ALT_GR);
    Keyboard.press(0xEC);
    Keyboard.releaseAll();
  }
}
```
For some special characters you can use Keyboard.press()  with corresponding codes by creating a lookup table or something else:
```
21 !
22 Ä
23 §
24 $
25 %
26 /
27 ä
28 )
29 =
2A (

2C ,
2D ß
2E .
2F -
 
3A Ö
3B ö
3C ;
3D ´
3E :
3F _
40 "

5B ü
5C #
5D +
5E &
5F ?
60 ^

7B Ü
7C '
7D *
7E °
```

For other keys you can use Keyboard.press() with existing definitions.