NicoHood/HID #97: Keyboard not recognised by OSX

RedLectroid, 2017-01-08:
I've noticed that when I use  this on OSX that the layout of the emulated keyboard is not recognised, so if you plug it into a mac, the Keyboard Assistant pops up and asks to you to press 'Z' on the new keyboard, which I obviously cannot do.  

I am using a US English keyboard on my MacBook.  It is updated to the latest version, macOS 10.12 Sierra.  

Oddly the default Arduino library is recognised by the MacBook and will happily type away, but it does not contain the LED detection that your library has, which my project desperately needs.

NicoHood, 2017-01-08:
That is weird because the lib has been working for a long time now. Did you try a different os?

RedLectroid, 2017-01-08:
It works fine in Linux and windows 7, 8.1 and 10. I don't have any older versions of osx to test, unfortunately, but it certainly doesn't work on the latest.

As I mentioned, it's weird the stock Arduino keyboard library works fine with osx, but yours doesn't.

NicoHood, 2017-01-08:
No idea, you might want to try to modify the report descriptor.

RedLectroid, 2017-01-08:
Could you direct me to where the report descriptor is and I can try? If it works I'll let you know.

NicoHood, 2017-01-08:
Wait. Which keyboard did you use? The Bootkeyboard or the normal keyboard?

The descriptors are in those files:
https://github.com/NicoHood/HID/blob/master/src/SingleReport/BootKeyboard.cpp

RedLectroid, 2017-01-08:
I was using the normal keyboard. I'll take a loot and see what I can do.

RedLectroid, 2017-01-11:
After looking into it and trying a variety of things I can confirm I do not understand report descriptors.  I tried to modify the BootKeyboard.cpp so it more closely resembled the Keyboard.cpp file from the Arduino library, but I just can't get it right.  

I'll keep trying, because without it working on OSX, my project loses quite a bit of appeal.  I'll update this if I find a solution.  Thanks for your fast replies.

Pigeo, 2017-07-17:
For your information : when you insert any non-Apple keyboard into a computer running Mac OS X, "the Apple Keyboard Assistant pops up and asks to you to press 'Z' on the new keyboard, which I obviously cannot do." => just click on the red cross button in the upper left corner of the keyboard assistant's window in order to close this window and this will bypass the "press the Z key" step, and then your keyboard will function normally (skipping the Keyboard Assistant means OS X will not have any clue concerning the mapping of this USB keyboard, and in such cases, it will use the default keyboard mapping, which is US keyboard mapping, if I remember well)

Pigeo, 2017-07-17:
There's only 2 ways to disable this "Apple Keyboard Assistant pop up" : on a single machine, you can edit a preferences file and put some information in it in order to disable the keyboard assistant, but it's on a single machine by machine basis. Otherwise, if you don't want your arduino to trigger this "Apple Keyboard Assistant pop up", you have to make your arduino HID keyboard faking as if it was a genuine "Apple keyboard", e.g. put USB Vendor ID (VID) to 0x5ac (Apple Computer, Inc) and USB Product ID (PID) to any existing Apple keyboard product ID (such as 0x221 for example - which corresponds to Apple USB MB110S/A European ISO keyboard with Swedish mapping) and voilà, no more "Apple Keyboard Assistant pop up" :-)
When the USB Vendor ID is Apple (and only when it is Apple), the Product ID gives Mac OS X a hint about the keyboard layout mapping used by this keyboard, so you have to choose the right Product ID depending if you want to be identified as a QWERTY or an AZERTY or a QWERTZ, etc.

Source : https://forum.yubico.com/viewtopic.php?f=4&t=111
See also : https://en.wikipedia.org/wiki/Apple_Keyboard

Here are some Apple Keyboard Product IDs : 
0x221 = MB110S/A = Swedish, European ISO, with Num. Keypad
0x250 = MB110LL/A = same as above, but with french (AZERTY) layout (Model No. A1243)
0x021b = Internal keyboard of a MacBook Pro 2007, with European ISO French layout
0x0205 = Apple Pro Keyboard (black) with European ISO French layout (Model No. M7803)