NicoHood/HID #99: Correct method to use multiple HID devices?  (ie. BootKeyboard + System)

sofakng, 2017-02-16:
If I want to use the BootKeyboard HID device but also send System and/or Consumer keys, do I need to initialize all three classes in my setup?

```
void setup()
{
  BootKeyboard.begin();
  Consumer.begin();
  System.begin();
}

void loop()
{
  System.write(SYSTEM_WAKE_UP);
  delay(5000);
  BootKeyboard.write(KEY_ENTER);
  Consumer.write(MEDIA_PLAY_PAUSE);
}
```

Can I use multiple HID devices with the TeensyKeyboard as well?  (TeensyKeyboard + System + Consumer)

NicoHood, 2017-02-16:
Your code looks correct. Teensykeyboard is not supported anymore. Where did you get this information from?

sofakng, 2017-02-16:
Thanks for the reply.  The Wiki has a link of the side labeled "Teensy Keyboard".

Are you saying that the HID project doesn't support Teensy at all?

NicoHood, 2017-02-16:
Thanks. I have removed the entry, it was old.
Well you can use the arduino micro core for the teensy. The processor is the same, but the bootloader is different. So unless you have a deeper knowledge of the stuff it is incompatible, yes.

closed 2017-02-16 by NicoHood