NicoHood/HID #100: Uno v3

jeanroukas, 2017-03-28:
Hi,
When I upload my code in dfu USB with Hoodloader2 16u2, I can access with code to the led but not the input (like button, for exemple INPUT 2). Not sure how to use the library, do I need to use serial to communicate between the 2 chipset ?

NicoHood, 2017-03-28:
Yes, Please read the Hoodloader2 Wiki carefully.

closed 2017-03-28 by NicoHood

jeanroukas, 2017-03-28:
Thanks for the quick answer. I've been like 10 hour on it, so I must be confused... So this is not a library that make ATmega328 via 16u2 to USB HID ? It's just a library for the 16u2 and I have to build the bridge between the 2 chipsets ?

jeanroukas, 2017-03-28:
Also I have seen this exemple : [https://github.com/NicoHood/HoodLoader2/tree/master/avr/examples/HID-Bridge](https://github.com/NicoHood/HoodLoader2/tree/master/avr/examples/HID-Bridge)

NicoHood, 2017-03-29:
Hoodloader2 Is a bootloader that enables you the opportunity to flash code to the 16u2. This also includes USB stuff.

Now you have 2 parallel running microcontrollers. you can always reset into bootloader mode to flash the 328p. but if you want to do something with both of them you need to communicate via serial between them. example is in the wiki, but the rest you have to do on your own. There are also other protocols available for arduino that do that job like firmata or the other one mentioned on hackaday that only requires a single wire.