NicoHood/HID #102: BootKeyboard - Missing Keypresses

Wh1terat, 2017-04-04:
Hi,
(Apologies first as this maybe semi device/board specific.)

I've got a sketch running on a pro micro clone (5v/16mhz) with boot protocol support.
(confirmed by toggling the rx led if BootKeyboard.getProtocol() == HID_BOOT_PROTOCOL)

It's working; however it seems to miss key strokes (~20%) - but only when using boot protocol.
Same host (and port) in report protocol with the same write code works without fault.

It can miss any of the characters at any time; disconnecting and reconnecting the device does not produce the same dropped characters.
I did try and set the LSM bit but also no improvement.

Really scratching my head on this - I know USB is very time sensitive so would like to blame the board (crystal/oscillator) but that wouldn't explain why there are no issues when in report mode.

Any ideas (however crazy) would be much appreciated!

NicoHood, 2017-04-05:
so you have this problem within the OS? Which OS do you use?
Some implementations set a delay between each press (mooltipass) as it is a known issue. but i never had this issue. so i am wondering too, especially for boot protocol.

Wh1terat, 2017-04-06:
Within an OS (OSX or Linux) in report mode it works without issue; the issue is seen in the OSX EFI Environment when using boot protocol.

I did have a delay of 500ms between each key stroke - and I can see the tx led flash on the pro micro when it is meant to have sent the keystroke but it doesn't product a key.
I did also try with press, delay and release and still faced the problem.

I'm starting to wonder if there is any kind of requirement to send a ZLP after each report or something like this in boot protocol mode?

NicoHood, 2017-04-06:
Not that I know of. You can post a sketch to reproduce this problem and someone may test it. I will be away for 2 weeks now.

Wh1terat, 2017-04-06:
No problem (and thanks for the replies to date) - It's just one of those projects where I don't actually have a need for it; just want to understand how to do it :)

[EFI-4pin.txt](https://github.com/NicoHood/HID/files/902952/EFI-4pin.txt)

Wh1terat, 2017-04-11:
Hmm, definitely something specific to this library although I'm not sure where exactly.

I went back and patched in boot protocol support into the standard Keyboard and HID libraries and I don't see the problem.

Anywho that ultimately solves my issue - but an odd one nonetheless!

NicoHood, 2017-04-23:
Can you please post your patches of the official libs? the whole file and the diff for better view.

Wh1terat, 2017-04-27:
Bit hacky (e.g making protocol a public var) and can ignore "hidgetdesccnt" 
(I needed to know how many times get descriptor had been queried)

[keyb.patch.txt](https://github.com/NicoHood/HID/files/962982/keyb.patch.txt)

w-A-L-L-e, 2019-04-04:
So is this patch integrated in newer versions as I'm having the exact same issue with boot keyboard on os x?

NicoHood, 2019-04-05:
No it was not integrated yet. But you could open a PR, so we can better review the patch and I am happy to merge it.

w-A-L-L-e, 2019-04-10:
Will do, as soon as we get it working here. I applied the patch but it didn't solve it yet for us. We will work on it in the next weeks most likely because we need it for our projects. And of course we'll create a pull request when we find a fix.

NicoHood, 2019-07-05:
Inlining the patch, so it wont get deleted any time later:

```c
diff -Naur orig/HID.cpp patch/HID.cpp
--- orig/HID.cpp	2017-04-27 22:14:05.000000000 +0100
+++ patch/HID.cpp	2017-04-27 22:24:14.000000000 +0100
@@ -30,7 +30,7 @@
 {
 	*interfaceCount += 1; // uses 1
 	HIDDescriptor hidInterface = {
-		D_INTERFACE(pluggedInterface, 1, USB_DEVICE_CLASS_HUMAN_INTERFACE, HID_SUBCLASS_NONE, HID_PROTOCOL_NONE),
+		D_INTERFACE(pluggedInterface, 1, USB_DEVICE_CLASS_HUMAN_INTERFACE, HID_SUBCLASS_BOOT_INTERFACE, HID_PROTOCOL_KEYBOARD),
 		D_HIDREPORT(descriptorSize),
 		D_ENDPOINT(USB_ENDPOINT_IN(pluggedEndpoint), USB_ENDPOINT_TYPE_INTERRUPT, USB_EP_SIZE, 0x01)
 	};
@@ -59,6 +59,7 @@
 	// due to the USB specs, but Windows and Linux just assumes its in report mode.
 	protocol = HID_REPORT_PROTOCOL;
 	
+    hidgetdesccnt = hidgetdesccnt + 1;
 	return total;
 }
 
@@ -87,12 +88,9 @@
 }
 
 int HID_::SendReport(uint8_t id, const void* data, int len)
-{
-	auto ret = USB_Send(pluggedEndpoint, &id, 1);
-	if (ret < 0) return ret;
-	auto ret2 = USB_Send(pluggedEndpoint | TRANSFER_RELEASE, data, len);
-	if (ret2 < 0) return ret2;
-	return ret + ret2;
+{   
+	auto ret = USB_Send(pluggedEndpoint | TRANSFER_RELEASE, data, len);
+	return ret;
 }
 
 bool HID_::setup(USBSetup& setup)
@@ -111,11 +109,12 @@
 			return true;
 		}
 		if (request == HID_GET_PROTOCOL) {
-			// TODO: Send8(protocol);
+            USB_Send(pluggedEndpoint, &protocol, sizeof(protocol));
 			return true;
 		}
 		if (request == HID_GET_IDLE) {
-			// TODO: Send8(idle);
+            USB_Send(pluggedEndpoint, &idle, sizeof(idle));
+            return true;
 		}
 	}
 
diff -Naur orig/HID.h patch/HID.h
--- orig/HID.h	2017-04-27 22:14:13.000000000 +0100
+++ patch/HID.h	2017-04-27 22:26:12.000000000 +0100
@@ -64,8 +64,8 @@
   uint8_t len;      // 9
   uint8_t dtype;    // 0x21
   uint8_t addr;
-  uint8_t versionL; // 0x101
-  uint8_t versionH; // 0x101
+  uint8_t versionL; // 0x01
+  uint8_t versionH; // 0x11
   uint8_t country;
   uint8_t desctype; // 0x22 report
   uint8_t descLenL;
@@ -95,6 +95,8 @@
   int begin(void);
[... 95 lines cut ...]
 Keyboard_::Keyboard_(void) 
@@ -74,6 +76,11 @@
 {
 }
 
+int Keyboard_::getProtocol(void)
+{
+    return HID().protocol;
+}
+
 void Keyboard_::sendReport(KeyReport* keys)
 {
 	HID().SendReport(2,keys,sizeof(KeyReport));
diff -Naur orig/Keyboard.h patch/Keyboard.h
--- orig/Keyboard.h	2017-04-27 22:15:28.000000000 +0100
+++ patch/Keyboard.h	2017-04-27 22:27:59.000000000 +0100
@@ -92,6 +92,7 @@
   size_t press(uint8_t k);
   size_t release(uint8_t k);
   void releaseAll(void);
+  int getProtocol(void);
 };
 extern Keyboard_ Keyboard;
 
```

w-A-L-L-e, 2019-07-05:
I didn't get above patch to work/fix the issue on mac yet. However we found out it only happens when file vault is enabled. Then the login screen is a different one (before the os is booted as the drive is still encrypted). This explains the different behaviour. And by googling a lot I found a different project with the similar issue and their patch for it (putting it here as it might be useful): 

https://github.com/kiibohd/controller/commit/8fbc8d535f6fd858fb549c4f85c83ee1791f77d7#diff-b5ce78718e4ff6a8178c2d7f7a925524

Hope it helps and maybe with some more time I can try again to make the above patch work...

Wh1terat, 2019-07-10:
There's some specific crud in my patch that isn't required (i.e exposing protocol and the number of times get descriptor has been called).

Fundamental changes:
- Add HID_SUBCLASS_BOOT_INTERFACE to the HID descriptor  
- Change the USB HID version (bcdUSB) from 1.01 to 1.11 (Can't remember exactly why I did this)
- Implement HID_GET_PROTOCOL and HID_GET_IDLE

The link you provided above was to implement HID_GET_PROTOCOL - that project also sets a USB HID version (bcdUSB) of 2.00 or 2.01.

So I suspect it requires HID_GET_PROTOCOL support and _potentially_ a bcdUSB version =>1.11

w-A-L-L-e, 2019-07-10:
Thanks for the extra info, will try to apply the changes this weekend and test it on a macbook login screen with filevault enabled.

Wh1terat, 2019-07-10:
Additionally, the report descriptor was not compliant with USB spec for boot keyboards which is why that was updated - again, been a while so can't remember if that was needed or not.

> The HID Subclass 1 defines two descriptors for Boot Devices. Devices may
append additional data to these boot reports, but the first 8 bytes of keyboard
reports

`https://www.usb.org/sites/default/files/documents/hid1_11.pdf`

bitboy85, 2020-07-09:
@Wh1terat Do you have a working version of your patch compatible with the recent IDE version? I also search for a working boot keyboard but sadly wasn't able to get your patch compiling.

Wh1terat, 2020-07-13:
@bitboy85 Afraid not, in all honesty I've probably not used an arduino for ~2 years.
That said, everything that's posted above should be enough to get you in the right direction.

bitboy85, 2020-07-14:
@Wh1terat 
I was able to implement some of your changes, but its still not detected. In my KVM Switch Thread i've documented where i got so far. I guess a compliant boot keyboard implementation requires a lot of changes in core files, for example the boot descriptors should be 8 bytes but it seems the current implementation defaults to 64...
Page 59

> Appendix B: Boot Interface
> Descriptors
> The HID Subclass 1 defines two descriptors for Boot Devices. Devices may
> append additional data to these boot reports, but the first 8 bytes of keyboard
> reports and the first 3 bytes of mouse reports must conform to the format defined
> by the Boot Report descriptor in order for the data to be correctly interpreted by
> the BIOS. The report may not exceed 8 bytes in length.