NicoHood/HID #105: Blink1 WS2812 led support

baruch, 2017-04-29:
I'm working on a device to create a volume knob with WS2812 led and want to use the Blink1 protocol (https://github.com/todbot/blink1) to integrate with whatever program is needed. From a quick look I can't find how to add a new hid report into the system and get the callbacks needed.

I would appreciate pointers into what needs to do and what to look at in order to achieve this. I will also happily contribute the Blink1 code back in case it will be useful for others.

NicoHood, 2017-05-02:
You'd have to read the arduino documents about implementing a pluggable hid module. See their wiki page on github. My project has nothing to do with this, also I dont know about blink1 and dont have any time to help now. I am really sorry. If you have any progress though I'd like to see what you've done, it sounds interesting. :)
Good Luck!

closed 2017-05-02 by NicoHood

baruch, 2017-05-05:
I've implement it at https://github.com/baruch/VolumeKnobBlink1
This is an ugly hack but it works.

To make it cleaner the HID interface would need to provide callbacks that can read data and not just send data. Presumably that's why your BootKeyboard doesn't work in the multi instance version.

NicoHood, 2017-05-06:
What exactly do you mean? #102 ?