NicoHood/HID #106: Is there a support for horizontal mouse wheel?

NMHoang05, 2017-05-04:
Hello!
My current project is to build a hand-wired one-handed keyboard with Arduino Pro Micro and it still has some spare pins and I thought of attaching a joystick to it.

my preference is to use the joystick as mouse wheel scroll, vertically and **horizontally**.
but I don't know if this library support the mouse wheel or should I look elsewhere?

by the way, I'm on Window 10 and horizontal scroll support is horrible. ("non-existence", I may say)

Thank you,
Hoang.

NicoHood, 2017-05-04:
I dont know if its even technically possible. You might want to give me some reference implementations or look at the usb hid docs.

NMHoang05, 2017-05-04:
hmm... right now I have not started coding yet (I will reuse code from my previous hand-wired keyboard)

I will search around and I will get back here if I find anything.

closed 2017-05-04 by NMHoang05