NicoHood/HID #108: Leonardo Bootloader Corruption when I use RawHID.begin

unicorn79, 2017-05-22:
Good afternoon! I use this sketch to work with boot keyboard, mouse and USBHID (i have arduino-ide-1.8.0-windows):

`#include "HID-Project.h"
#include <avr/wdt.h>

uint8_t rawhidData[255];

void setup() {
 
  // initialize mouse and keyboard control:
  BootKeyboard.begin();
  BootMouse.begin();
  RawHID.begin(rawhidData, sizeof(rawhidData));
  wdt_enable (WDTO_2S);
}

void loop() {
	wdt_reset();

}`

it works, after burning, but when i load next sketch (even the same)  to my Arduino Leonardo (clone), it seems to corrupt my Leonardo Bootloader or some kind.
So if i try to Load this sketch once again by pressing reset button, no "Arduino Leonardo (COM$$)" appears, so sketch loading cannot be completed.
When i reload leonardo Bootloader with USBASP (from IDE - "Load Bootloader"), and try to load sketch  , where is excluded RawHID.begin(rawhidData, sizeof(rawhidData));),  the Leonardo bootloader works fine and seems to be not corrupted, it appears good when i press "reset" button...

 `#include "HID-Project.h"
#include <avr/wdt.h>

uint8_t rawhidData[255];

void setup() {
 
  // initialize mouse and keyboard control:
  BootKeyboard.begin();
  BootMouse.begin();
  //RawHID.begin(rawhidData, sizeof(rawhidData));
  wdt_enable (WDTO_2S);
}

void loop() {
	wdt_reset();

}`

Any ideas? Thanks!!

NicoHood, 2017-05-22:
I dont really understand your issue, no do i understand why you are using the watchdog?

The bootloader should **always** run if you press enter and then plug in the device.

unicorn79, 2017-05-24:
I use the standart arduino bootloader from IDE (Tools->write loader), not a Hoodloader2. So i loading my sketch to my Arduino Leonardo, and it works fine. As you can see it is a 2 USB HID Devices  At the same time (1 USB HID Keyboard and 1 USB HID Mouse).. The code:

```cpp
#include "HID-Project.h"

uint8_t rawhidData[255];

void setup() {
 
// initialize mouse and keyboard control:
  BootKeyboard.begin();
  BootMouse.begin();
}

void loop() {

}
```

After that i press the reset button, wait the Arduino Loader Appears in COM-ports (its OK), and then load next sketch with a 3 device At the same time (1 USB HID Keyboard, 1 USB HID Mouse, 1 Custom USB HID Device). The code is following:

```cpp
#include "HID-Project.h"

uint8_t rawhidData[255];

void setup() {
 
// initialize mouse and keyboard control:
  BootKeyboard.begin();
  BootMouse.begin();
  RawHID.begin(rawhidData, sizeof(rawhidData));
}

void loop() {

}
```

it seems to loading OK in Leonardo, and works fine, but All my further attempts to load the any other sketches fail, because the Arduino loader does not work any more. It seems to be corrupted, when i load more than 2 devices in sketch at the same time. As you can see in the previous code - there are 3 USB HID devices in a row.  I think this is the reason for the corruption of the Arduino bootloader. The Arduino Bootloader simply not appears in COM-ports after i press Reset button.

Did anyone else have such problems with the bootloader, when you load sketch with 3 devices that used at the same time (mouse + keyboard + custom USB HID)? 

I will be grateful for your advice!

Thanks!

PS: Sorry for my English ((

NicoHood, 2017-05-24:
Do you use windows? I currently have no device and time to test. Someone else would need to help you. Or you should open a bug at the arduino project. Because this should never kill a bootloader. I am really confused whats happening.

unicorn79, 2017-05-25:
I use Windows 7 x64 OS, and I'm surprised that no one before this incident has encountered such a problem. I will try myself to find out the cause of such an undesirable effect, and, in case of luck,  I'll try to describe the solution there.

unicorn79, 2017-05-27:
For those, who will be getting a similar problem: Simultaneous use of three devices in one Arduino Leonardo - 1) Boot-keyboard  2) Boot-mouse 3) Custom USB HID device
It was determined by the test method that the function RESET is working incorrectly by Pressing the Reset button on the Leonardo board. However, when you unplug the USB-connector of Leonardo and connect in back - the Bootloader appears correctly, which makes it possible to load the sketch further.

NicoHood, 2017-05-27:
@unicorn79 Thanks for the feedback. I think I reported that issue somewhere but I can only find this as reference:
https://github.com/arduino/Arduino/issues/3756

Can you maybe try HoodLoader2? It wont cause you any additional trouble and might fix your issue. Or you can try to compile the LUFA CDC bootloader with my PR (arduino fix) applied: https://github.com/abcminiuser/lufa/pull/86