NicoHood/HID #109: Leonardo Compile error

JazzyManSerg, 2017-06-02:
Arduino: 1.6.7 (Windows 7), Плата:"Arduino Leonardo"

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\Gamepad.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/USBAPI.h:89:1: error: expected class-name before '{' token

 {

 ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/USBAPI.h:105:8: error: 'Print' has not been declared

  using Print::write; // pull in write(str) and write(buf, size) from Print

        ^

exit status 1
Compile error

JazzyManSerg, 2017-06-02:
This error on Arduino 1.6.7 when I compile Gamepad.h

NicoHood, 2017-06-02:
Please use the latest IDE

closed 2017-06-02 by NicoHood