NicoHood/HID #111: Can't build Example, HID.h not found

harisjmd, 2017-06-07:
```
15:54:30 **** Build of configuration Release for project sadas ****
"C:\\eclipse\\cpp-neon\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../sadas.cpp'
'Starting C++ compile'
"C:\eclipse\cpp-neon\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\eclipse\cpp-neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\eclipse\cpp-neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\leonardo" -I"C:\eclipse\cpp-neon\arduinoPlugin\libraries\HID-Project\2.4.4" -I"C:\eclipse\cpp-neon\arduinoPlugin\libraries\HID-Project\2.4.4\src" -MMD -MP -MF"sadas.cpp.d" -MT"sadas.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../sadas.cpp"  -o  "sadas.cpp.o"
In file included from C:\eclipse\cpp-neon\arduinoPlugin\libraries\HID-Project\2.4.4\src/HID-Project.h:39:0,
                 from ../sadas.h:11,
                 from ../sadas.cpp:2:
C:\eclipse\cpp-neon\arduinoPlugin\libraries\HID-Project\2.4.4\src/SingleReport/SingleAbsoluteMouse.h:29:17: fatal error: HID.h: No such file or directory
 #include "HID.h"
                 ^
compilation terminated.
subdir.mk:18: recipe for target 'sadas.cpp.o' failed
make: *** [sadas.cpp.o] Error 1

15:54:30 Build Finished (took 505ms)
```

Using this example:

```
#include "Arduino.h"

#include "HID-Project.h"

const int pinLed = LED_BUILTIN;
const int pinButton = 2;

void setup() {
  pinMode(pinLed, OUTPUT);
  pinMode(pinButton, INPUT_PULLUP);

  // Sends a clean report to the host. This is important on any Arduino type.
  BootKeyboard.begin();
}

void loop() {
  // Update Led equal to the caps lock state.
  // Keep in mind that on a 16u2 and Arduino Micro HIGH and LOW for TX/RX Leds are inverted.
  if (BootKeyboard.getLeds() & LED_CAPS_LOCK)
    digitalWrite(pinLed, HIGH);
  else
    digitalWrite(pinLed, LOW);

  // Trigger caps lock manually via button
  if (!digitalRead(pinButton)) {
    BootKeyboard.write(KEY_CAPS_LOCK);

    // Simple debounce
    delay(300);
  }
}

```

I can't build any example using arduino plugin for eclipse, checked also with visual micro for Visual Studio but i get the same error.. Builds ok with arduino ide 1.8.2. Any suggestions? Anyone with the same issue? 

Thanks in advance.

NicoHood, 2017-06-07:
I dont know what the plugin does, you have to use an USB MCU like the 32u4 (leonardo/micro)

harisjmd, 2017-06-07:
Yeah, I am using 32u4 with leonardo bootloader..The thing is that other libraries work with the plugin only with your library i get the above error..

NicoHood, 2017-06-07:
Did you try the official arduino keyboards?

harisjmd, 2017-06-09:
Yes they are compiling just fine
```
19:57:27 **** Build of configuration Release for project sad ****
"C:\\s\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sad.cpp'
'Starting C++ compile'
"C:\s\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\leonardo" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" -I"C:\s\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1\src" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID\src" -MMD -MP -MF"sad.cpp.d" -MT"sad.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sad.cpp"  -o  "sad.cpp.o"
'Finished building: ..\sad.cpp'
' '
'Building file: C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\utility\twi.c'
'Starting C compile'
"C:\s\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\leonardo" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" -I"C:\s\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1\src" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID\src" -MMD -MP -MF"libraries\Wire\src\utility\twi.c.d" -MT"libraries\Wire\src\utility\twi.c.o" -D__IN_ECLIPSE__=1 "C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\utility\twi.c"  -o  "libraries\Wire\src\utility\twi.c.o"
'Finished building: C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\utility\twi.c'
' '
'Building file: C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\Wire.cpp'
'Starting C++ compile'
"C:\s\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\leonardo" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" -I"C:\s\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1\src" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID\src" -MMD -MP -MF"libraries\Wire\src\Wire.cpp.d" -MT"libraries\Wire\src\Wire.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\Wire.cpp"  -o  "libraries\Wire\src\Wire.cpp.o"
'Finished building: C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src\Wire.cpp'
' '
'Building file: C:\s\arduinoPlugin\libraries\Keyboard\1.0.1\src\Keyboard.cpp'
'Starting C++ compile'
"C:\s\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\""  -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\leonardo" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\Wire\src" -I"C:\s\arduinoPlugin\libraries\PubSubClient\2.6.0" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1" -I"C:\s\arduinoPlugin\libraries\Keyboard\1.0.1\src" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID" -I"C:\s\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\libraries\HID\src" -MMD -MP -MF"libraries\Keyboard\sr [... cut ...]
  // read the pushbutton:
  int buttonState = digitalRead(buttonPin);
  // if the button state has changed,
  if ((buttonState != previousButtonState)
      // and it's currently pressed:
      && (buttonState == HIGH)) {
    // increment the button counter
    counter++;
    // type out a message
    Keyboard.print("You pressed the button ");
    Keyboard.print(counter);
    Keyboard.println(" times.");
  }
  // save the current button state for comparison next time:
  previousButtonState = buttonState;
}

```

numist, 2018-07-28:
Strange that the example would compile because I just downloaded 1.8.5 and `HID.h` is gone, which breaks any project that includes `Keyboard/src/Keyboard.h` on my machine.

nodoubtman, 2018-12-01:
Hello World.

I have solved the problem by doing this:

#include <HID.h>

the <> at the beginning of the file is very important.
Have a good night.
Marc.

closed 2022-02-23 by NicoHood