NicoHood/HID #112: RawHID doesn't work on atmega90usb162

wstrzalka, 2017-06-26:
Hi
    I'm trying to use RawHID on my prototype board with 90usb162 chip and it doesn't seems to work. 
    I'd compiled simplified version of the RawHID example and burned into 90usb162 and getting Unknown device in Windows Device Manager + nasty tray popup of malfunctioning device. Same code compiled & burned for 32u4 works fine there.
   I'm sure the hardware is OK as when I do try it with the Gamepad sample everything works fine. It's only the combination of RawHID & 90usb162 not working (and unfortunately that's what I need :) ) 
  I'm using Arduino 1.8.2 stack to build it with the board set to Hoodloader2 at90usb162.
 
  Any thoughts ?

NicoHood, 2017-06-27:
A similar issue was reported multiple times. I currently dont have the time nor the hardware to test. I am sorry.
https://github.com/NicoHood/HID/issues/110

wstrzalka, 2017-06-27:
OK - I think I found the root cause. It's as trivial as ..... out of memory. I found out the bearer buffer size (the array passed to the RawHID.begin() ) for the sample I was running to be 227 bytes running stable while 228 not working. But when I set 227 and added few bytes worth variables - bang - not working again.

I'm not really sure how this buffer size affects application (does it limits the maximum number of bytes I can send/receive in one shot or it's just performance thing??) but changing the example to be more conservative  (64 bytes??) and putting a comment regarding it's size above would save people a lot of time and frustration. 

Best regards 
   Wojtek

closed 2017-07-17 by wstrzalka