NicoHood/HID #113: MIDIUSB does not register as a MIDI device

canadaduane, 2017-07-13:
I'm not sure where to leave this, since old HoodLoader2 comments point here, and the wiki link here points to MIDIUSB. Any help is appreciated.

I'm trying to turn my Arduino Uno into a MIDI device (for experimental data transfer, not for MIDI itself per se). I've loaded HoodLoader2 onto the 16u2, and I'm running [this sample code](https://github.com/arduino-libraries/MIDIUSB/blob/master/examples/MIDIUSB_read/MIDIUSB_read.ino) from the MIDIUSB project (after installing MIDIUSB as a library).

What I'm expecting to see is the Arduino register itself as a MIDI USB device. But it remains unchanged:
- Mac OS X's `System Information` app identifies it as "HoodLoader2 16u2" with no serial number
- the port is `/dev/cu.usbmodem1411 (HoodLoader2 16u2)` in the Arduino IDE
- MIDI apps don't recognize the Arduino as being a MIDI device

When I run the [Gamepad example code](https://github.com/NicoHood/HID/blob/master/examples/Gamepad/Gamepad.ino), it behaves more like what I expect:
- `System Information` identifies the USB as having `Serial Number` "HIDAF"
- the port becomes `/dev/cu.usbmodemHIDA1 (HoodLoader2 16u2)` in the Arduino IDE
- game apps recognize the Arduino as a Gamepad

NicoHood, 2017-07-14:
It wont work, as the 16u2 only has 4 enpoints and midi uses 2 while serial uses 3. You need to write your own lufa application to only use midi. There is no way to remove the serial inside the arduino core.

closed 2017-07-14 by NicoHood

maddanio, 2021-02-25:
so this means there is no way to setup the uno as a midi device using hoodloader2?

NicoHood, 2021-02-25:
It is possible, but not with the default arduino stack. You either need to disable the CDC Serial or write your own firmware using lufa or other solutions.

maddanio, 2021-02-25:
how would i go about that first solution? an what does it mean? that i dont have serial monitor anymore? can I still talk between the cores in a straightforward way?

maddanio, 2021-02-25:
I am not really up to writing firmware :)

maddanio, 2021-02-25:
otherwise I will use other firmware, i just like a lot that hoodloader2 allows me to patch both cores via usb, meaning i dont have to disassemble my gadget to hack it :)

NicoHood, 2021-02-25:
You cannot use the serial monitor for the 16u2 anymore then. In bootloader mode you can still debug the 328 though. The patch to disable cdc serial can be found here:
https://github.com/arduino/Arduino/issues/6387#issuecomment-309363314

maddanio, 2021-02-25:
Ok, would i still be able to program both cores via usb then?

maddanio, 2021-02-27:
sorry for picking you brain like that, but I am still looking for the jackpot solution: having the arduino work as both a midi host and midi device (both over usb, using its onboard port and a usb host shield) while still being able to fully program it (both cores) without having to disassemble my gadget, and I guess this would also be quite nice for others. but if its not possible i'll have to go with what works :).

NicoHood, 2021-02-27:
Just use a 32u4, like a leonardo

maddanio, 2021-02-27:
Ah, ok, so thats basically a fully integrated single core arduino? They seem to be mich cheaper so i assume slower? But i dont need much compute, so yah, cool!