NicoHood/HID #114: Arduino Leonardo Micro (ATMEGA32U4): Keyboard volume does not work on Windows

codebeat-nl, 2017-07-18:
```
if( <buttonpressed> )
 { Keyboard.write( HID_KEYBOARD_VOLUME_DOWN ); }
if( <anotherbuttonpressed> )
 { Keyboard.write( HID_KEYBOARD_VOLUME_UP ); }
delay(10);
.......................
```

It does work on Android but not on Windows. Test it on Windows 7 and Windows 10.  Is there a fix for this?

NicoHood, 2017-07-18:
Its not working on Windows because Windows is Windows.
They did not implement it, like they did with many things.
It should work on Linux and maybe on IOS too.

closed 2017-07-18 by NicoHood

codebeat-nl, 2017-07-18:
Yes, is still working when you use the Consumer class instead of the Keyboard class. I don't know what the difference is between them?

NicoHood, 2017-07-18:
The consumer class is made for such things. Audio, Media keys, browser button etc. Thats the normal way to do it. You can also include it inside a keyboard descriptor, but windows just ignores that. For windows a keyboard only has "normal" keys, no consumer keys. Ask micro$$$$, not me about this issue. You cant get this fixed.