NicoHood/HID #115: HID-Project.h on Pro Micro Leonardo: HID Project can only be used with an USB MCU.

codebeat-nl, 2017-07-18:
On Pro Micro (Leonardo):
```
#if !defined(USBCON)
#error HID Project can only be used with an USB MCU.
#endif
```
Will triggered when using HID-Project with the AdaFruit NeoPixel library. Run the demo's and working fine without this error. So maybe this is not a reliable way to test capability.

NicoHood, 2017-07-18:
I dont know which classes you use. Can you give me more information about this? Otherwise this seems invalid to me.

codebeat-nl, 2017-07-18:
If you try to compile (Target: Arduino Leonardo) this code (in zip format, unzip it to project folder):

https://drive.google.com/file/d/0B2l-eQoHefcVRE9LRS1wTHFocG8/view?usp=sharing

You will get the error.  Maybe it is because I include the file in a class of mine? I use it as a part of my class. When I test for USBCON myself it is there, but when your code try to detect it, it is not there. Weird. I know the compiler can do sometimes strange things because of the build method that is used, it is not completely error free.  I'm using Arduino 1.8.2. Or maybe there is another reason? Please let me know if you get the same error. Thanks for the help.

NicoHood, 2017-07-18:
I aint got time to try this out. I recommend the arduino irc chat on freenode. They can help you, they usually know my library. They possibly use it more than me. I am really sorry. If you ping me at a later date, I might have time to revisit all those newer issues and try them out myself. Sorry.

bricewge, 2017-07-18:
I got the same issue with the same library and the same board. Commenting the 3 lines is a functional  work around.

Following is the code which triggered it:
``` arduino
#include "HID-Project.h"

void setup() { BootKeyboard.begin(); }

void loop() {
  delay(1000);
  BootKeyboard.write(KEY_ENTER);
}
```

JJablon, 2017-07-22:
I dont know what exactly problem did u have, but it helped,when i commented these lines in main .h file:
//#include "SingleReport/RawHID.h"
//#include "SingleReport/BootKeyboard.h"
//#include "MultiReport/ImprovedKeyboard.h"
//#include "SingleReport/SingleNKROKeyboard.h"
//#include "MultiReport/NKROKeyboard.h"

bricewge, 2017-08-04:
Actually with `platformio` including `Arduino.h` works for me.

johnfg10, 2018-12-18:
Also had this issue with the arduino IDE using platformio seems to of fixed it 👍