NicoHood/HID #119: feedback regarding smartphones

sindy39, 2017-09-26:
Referring to

> Could work with smartphones (feedback appreciated)

on the wiki.

Not sure whether the feedback is still interesting two years later, but **yes, it works well with Android smartphones**.

However, I've hit an issue which is most likely a Sony one but one never knows: if I "press" and "release" a NumLock, everything works fine except the LED state, as the CapsLock indicator comes on together with the NumLock one. When I "press" and "release" NumLock again, the NumLock indicator goes off but the CapsLock remains on. To be clear, this misbehaviour only affects the indicator, not the actual state of CapsLock.

To align CapsLock indicator state with the actual state, you have to "press" and "release" CapsLock twice. Processing of CapsLock and ScrollLock keys has no side effects.

The above only happens with my Sony Xperia Z2; an Alcatel phone I got at hand doesn't exhibit this behaviour, nor does a Lenovo notebook with Win10.

I'm not deep enough into USB HID protocol to be sure whether it is a clear bug at Sony side or whether there are several ways how the host may communicate the indicator state to the USB function and the library only supports one of them.

Pavel

NicoHood, 2017-09-26:
Hi,
thank you very much for the feedback. Maybe you could test with a real keyboard, it should have the same issue then.

Can you please also name us the androird version, you are using?

sindy39, 2017-09-26:
> Maybe you could test with a real keyboard, it should have the same issue then.

I actually did but forgot to write it in the first post. Yes, the behaviour is the same, but the keyboard is more than 10 years old.

Next week I'll capture the communication using a USB analyzer to be 100% sure what's running over the wire.

The affected phone reports Android version 6.0.1.

A new observation since the first post is that I was wrong when saying that CapsLock processing behaves normally, sending CapsLock while ScrollLock indicator is on switches the latter off. Not that I'd have any idea what ScrollLock might be good for on a smartphone.