NicoHood/HID #120: DefaultKeyboardAPI::set can leave KeyReport in a bad state

SteveBenz, 2017-10-22:
My PS2->USB keyboard translator manages to highlight this problem, which has been around for a while.

If you do:

press(key1)
press (key2)
release(key1)
press(key2)

Then DefaultKeyboarAPI::set fails to grok that key2 is already in the report, and causes key2 to appear in the key report twice.  So the next call to release(key2) will not work.

SteveBenz, 2017-10-22:
A pull request will follow shortly...

NicoHood, 2017-10-23:
Dont we need to check/fix the pressing code then? Only add a key, if its not in the list. Your code would also fix the issue, but from my point of view it would be better to fix it in the press() code, not release(). Correct me if I am wrong.

The code would work like this:
Search an empty slot, remember the number. Still check for the same key, abort if the key was already found. If not, save the key in the first empty slot.

SteveBenz, 2017-10-24:
There are two good ways to fix it.

1. You could make all the code paths look at all 8 bytes of the record (what you're proposing)
2. You could make all the code paths look at up to 8 bytes, stopping at the first empty entry (will result in the fastest code)

My change enables the second choice, but doesn't implement it, because I've always felt like the best policy in creating a first pull request into somebody else's codebase is to start out with the minimal change, rather than what I think is the best one (because there might be more I'm not aware of).

I'm happy to implement either of the two choices above; up to you.

SteveBenz, 2017-10-28:
I updated it to always keep the list without duplicates or 0's in the middle.  Basically, this means that the routine's speed depends on the number of keys pressed, which is good since most of the time that number is zero or one.

Tests are here:

https://github.com/SteveBenz/HidKeyboardSetTests

SteveBenz, 2017-11-04:
Not sure if you're busy with other things or what.  This is an important issue for any keyboard-based project.  Imagine a keyboard being used, say, for a WASD-based game - there will be plenty of times when a key will be held down while another key is being repeatedly pressed and released.  I found the issue in my application, which is a tweakable PS2-to-usb keyboard adapter.  There it primarily showed up because the PS2 keyboard has that auto-repeat feature, so it's reproducible just by holding down a pair of keys and releasing them in the order you pressed them.  ('a'<hold>'s'<hold> release a, release s, and now 's' is a dead key).

If you really wanted to follow the approach I outlined above, I created a second pull request that goes that road, but I think the first proposal is better just because it's fewer machine cycles.

NicoHood, 2017-11-04:
I got your PR on my todo list, I will try to review within the next week. Thanks so much for the help, you are all right. This is something that needs to get fixed quickly!

SteveBenz, 2017-11-07:
BTW, after we get the right pull request in good shape, don't merge it right away - I use the code on my main keyboard at work, so I can give it a pretty good shakedown over a period of a couple days to make sure there are no surprises.

NicoHood, 2018-01-31:
This is how I've reimplemented the code in my bare avr core:
https://github.com/NicoHood/avr/blob/master/lib/USB_KEYBOARD/src/usb_keyboard.c#L242

SteveBenz, 2018-02-04:
I must be missing something, but on inspection it sure looks like it's broken the same way it was before.  Splitting things out into two functions makes things a bit nicer, but it's still got the same flaw in it.

commit 78d325e1 (2018-09-15): fix: #120 bad keyreport issue
closed 2018-09-15 by NicoHood