NicoHood/HID #125: How to change the device name(HID)

starock, 2017-11-09:
I want to make a keyboard with HID porject. And show the device name(Arduino Leonardo) with what I want. Just like "MyKeyboard".

HID-project can do that?

BDanAnd, 2017-11-09:
I did this by editing the file "(ArduinoIDERootDir)\hardware\arduino\avr\boards.txt". Just copy the description block of Arduino Leonardo, change the "leonardo." prefix to your,  change the "<your_prefix.name" (for Arduino IDE menu) and then change the "usb_product". I checked on the Arduino Micro.

starock, 2017-11-10:
> I did this by editing the file "(ArduinoIDERootDir)\hardware\arduino\avr\boards.txt". Just copy the description block of Arduino Leonardo, change the "leonardo." prefix to your, change the "<your_prefix>.name" (for Arduino IDE menu) and then change the "usb_product". I checked on the Arduino Micro.

Thank you so much, I did it! hahaha~

closed 2017-11-10 by NicoHood