NicoHood/HID #129: Can't use my usual keycodes with included HID-* libraries

funcMYXOMOP, 2018-01-21:
I wanted to use some media keys in my existing project, so I just downloaded and installed the NicoHood libraries. After compiling, I found that my ctrl, shift, alt and the menu keys I sent via keycodes (Keyboard.write(237) for examlpe) just stopped working. I would like to know a way to use ctrl, alt, shift, menu keys and media keys in one project.

NicoHood, 2018-01-21:
There is no reason why it should not work. I guess you are using the wrong API commands, please check the examples. You should not pass the values like this to the write() API, you need to send them as keycodes. In the example everything is explained.

funcMYXOMOP, 2018-01-21:
Thx for responding. But i still don't understand what i'm doing wrong. Maybe i must send keys through Consumer.press(.. commands instead of Keyboard.press(.. ? But i don't know correct modificators names.

funcMYXOMOP, 2018-01-21:
Looked in ImprovedKeyboard.ino example. Can i reach my goal with Keyboard.write(KeyboardKeycode(n)), where n will be my keycodes i used earlier?

NicoHood, 2018-01-21:
Keyboard.write(KEY_LEFT_SHIFT);

https://github.com/NicoHood/HID/blob/master/src/HID-APIs/ImprovedKeylayouts.h#L258

closed 2018-01-21 by NicoHood