NicoHood/HID #130: HID Serial for feature reports

wormyrocks, 2018-03-19:
Hi Nico,

I've made good use of this library for a variety of projects, so thanks a lot for building it!

I am trying to implement a device that normally functions as a 5-button joystick, but has bidirectional serial communication for setting calibration values in EEPROM. Something exactly the same as [this](https://github.com/obdev/v-usb/blob/master/examples/hid-data/firmware/main.c), or similar to [this](http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/#comment-173). 

So far, I have a descriptor that shows up as a joystick, with an extra 32-byte feature report, but I'm not sure how to interface with it using HIDapi, or how to modify RawHID.cpp to handle requests from a host PC. It seems like at a minimum I need to modify RawHID_.setup, and change the PluggableUSBModule inheritance to include a non-null feature report and feature length? I am attaching my modified RawHID.h, perhaps I need to change it around [here](https://gist.github.com/wormyrocks/3976de8d11ef7f37341c44e68c67ac85#file-rawhid-cpp-L56)?

Thanks for any help you might be able to offer, I will look deeper into the other libraries in the meantime.

NicoHood, 2018-03-20:
I have a feature report API here:
https://github.com/NicoHood/HID/blob/master/src/SingleReport/BootKeyboard.h#L42

However I have no idea how I intended to use it :D Sorry.
On the host site I am not sure if HIDAPI can handle feature reports, but I think it can. Otherwise you need to use USBAPI

mcuee, 2023-06-17:
For host side, hidapitester is a good tool to test HID device.
https://github.com/todbot/hidapitester